OpenEMS/openems #7: Integrate weather data for forecasts

sfeilmeier, 2017-06-12:
### Bug Report or Feature Request (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [x] feature request
```

### Desired functionality.

There should be Bridge/Device/Channels to provide weather data information to the system.
Possible providers could be: 
- https://www.meteoblue.com/de/pointplus
- https://darksky.net/dev/docs/forecast

KyleMicrocare, 2019-09-03:
This is also functionality we've discussed at Microcare for predicting PV output within a certain confidence interval. The stats can get quite complex very quickly, especially when factoring in seasonality, PV panel facing direction, PV panel soiling, etc.

The Norwegian Meteorological Institute actually has some very comprehensive APIs available:
https://api.met.no/

sfeilmeier, 2019-09-10:
Thanks Kyle. To get better visibility of this topic, I moved it to the OpenEMS Community forum (https://community.openems.io/t/integrate-weather-data-for-forecasts/123). Please find my comments on current development there; I will close the issue here.

closed 2019-09-10 by sfeilmeier