OpenEMS/openems #31: i18n

ebakir, 2017-06-30:
### Bug Report or Feature Request (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [x ] feature request
```

### Desired functionality.

Enable translations of the application. See https://angular.io/docs/ts/latest/cookbook/i18n.html for details

ebakir, 2017-06-30:
Translation of strings in code is not supported at the moment.
See https://github.com/angular/angular/issues/11405
Possible workaround: https://github.com/angular/angular/issues/11405#issuecomment-295151768

ebakir, 2017-06-30:
mybe try this tutorial:
http://blog.danieleghidoli.it/2017/01/15/i18n-angular-cli-aot/#

German tutorial at Heise:
https://www.heise.de/developer/artikel/Internationalisierung-mit-Angular-Bordmitteln-3703621.html#mobile_detect_force_desktop

Second part of the tutorial at Heise:
https://www.heise.de/developer/artikel/Internationalisierung-fuer-Angular-Teil-2-ngx-translate-3722374.html?wt_ref=https%253A%252F%252Fwww.google.com%252F&wt_t=1495666349133

ebakir, 2017-06-30:
https://github.com/OpenEMS/openems-gui/commit/76980009d8f41c5b0605ce02a5ce912377250946

sfeilmeier, 2017-07-26:
@ebakir Please continue translation of all terms that are in the monitoring. Some terms (like "Angemeldet als Benutzer "guest"") are coming from the Edge component and need to be handled separately. 

Please also remove all "i18n" attributes still existing from earlier tries.

ebakir, 2017-08-01:
[https://github.com/OpenEMS/openems/commit/ffb35abd29a5e751058af8d762c127bbc6e2a56b](url)

ebakir, 2017-08-03:
https://github.com/OpenEMS/openems/commit/607c7c3ddb436079f9a37ba381f36fb0dcd86f86

ebakir, 2017-08-03:
ToDo: 
Load language automatically from odoo

ebakir, 2017-08-10:
https://github.com/OpenEMS/openems/commit/e5e71a7a4ea244956d29794f7205fe7cb3e67b94

ebakir, 2017-08-11:
https://github.com/OpenEMS/openems/commit/ab5e78e6099dfb47dc93543746fe33a07db43faf

closed 2017-08-28 by ebakir