OpenEMS/openems #64: No information about building the software

PlotlyUser2, 2018-08-03:
### Bug Report or Feature Request (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [x] feature request
```

### Bug description or desired functionality.

sfeilmeier, 2018-08-06:
Dear PlotlyUser2,

thanks for opening an issue. The current [documentation](https://github.com/OpenEMS/openems/blob/develop/doc/OpenEMS_EN.adoc) covers the Getting Started part. Were you able to follow that? 

It indeed misses the information on building e.g. a JAR-file currently. To build an executable "Fat-JAR" that can be executed outside of Eclipse IDE, simply open the "EdgeApp.bndrun" file in Eclipse and click the "Export" button next to "Run OSGi".

[image]

Is that information helpful for you?

sfeilmeier, 2018-08-07:
Hi,

- Is it working if you execute it inside Eclipse IDE?
- Which Java version are you using? ("java -version" in the console)
- Can you try to manually "Resolve" the EdgeApp.bndrun file using the button on the lower left?

PlotlyUser2, 2018-08-08:
Hello Stefan,

I got the issue. In fact I had java 10 on my PC installed, but in the .bndrun Execution environment I had Java 8. 

However, I wonder why this jar build cannot be supported within java 10...

sfeilmeier, 2018-08-08:
Hi. Glad you solved the problem. I have seen something similar at a colleagues machine where he had problems with Java 10. I was not yet able to solve the issue... it's strange. Also we didn't have the necessity as we use Java 8 with our productive devices.

Is the issue solved for you?

May I ask you about your plans with OpenEMS and if we can be of any assistance?

sfeilmeier, 2018-08-09:
Glad to hear that.

The 'ng serve' webserver is only meant for development purposes. It is not suitable for productive use. There you should compile the UI using "ng build" and serve that folder e.g. using NGINX using a configuration like this:
```
http {
        include mime.types;
        default_type application/octet-stream;
        sendfile on;
        keepalive_timeout 5;

        server {
                listen 80;
                root /usr/share/nginx/html/;
                rewrite_log on;

                location / {
                        rewrite ^ /m/$args last;
                }

                location /m/ {
                        index index.html;
                        try_files $uri$args $uri$args/ /index.html;
                }
        }
}
```

See the official Angular documentation for more information: https://angular.io/guide/deployment

I am closing this issue here as the original problem was solved. Please feel free to open another one for any upcoming problem.

closed 2018-08-09 by sfeilmeier