OpenEMS/openems #79: Error in connection establishment

PlotlyUser2, 2018-08-14:
### Bug Report or Feature Request (mark with an `x`)
```
- [x ] bug report -> please search issues before submitting
- [ ] feature request
```

Hi Stefan,

I've build the ui source with 

> 
ng build
> 

and serving the app with the basic configuration of the lighttpd webserver.
However, anytime I try to open the webpage I get this connection refused error:

[image]

What is going wrong here? Further, I've noticed that this issue was already reported, see #63...
PS: I'm on latest Ubuntu

sfeilmeier, 2018-08-14:
Hi,

can you verify that the "Controller Api Websocket" is running? See https://github.com/OpenEMS/openems/blob/develop/doc/OpenEMS_EN.adoc#23-run-openems-edge-and-start-simulator, paragraph "viii". It needs to be reachable on "ws://localhost:8085".

PlotlyUser2, 2018-08-14:
Yeah... that was a triple facepalm... I missed that option somehow... It's working

[image]

sfeilmeier, 2018-08-14:
Haha. I'm glad it's working. :laughing: :+1:

closed 2018-08-14 by sfeilmeier