OpenEMS/openems #82: [modbus rtu serial] Invalid syntax in target property for dependency Modbus to

PlotlyUser2, 2018-08-17:
### Bug Report or Feature Request (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
```
Hi Stefan,

I try to get some voltage and current values by using the modbus serial (RS485) rtu and the abb b23/b24 meter see: https://library.e.abb.com/public/17d9ba907144b09fc1257b98004c513a/2CMC485003M0201_A_en_B23_B24_User_Manual.pdf.pdf. 

So far I've followed your tutorial for programming a modbus interface (4.3.4. Implementing a Device). However, when I configure the serial modbus interface via Apache Felix Web Console I get this error:

`ERROR [io.openems.edge.meter.b23.abb ] [Meter.ABB.b24(17)] Invalid syntax in target property for dependency Modbus to`

To simplify things, I've tried to read only the first two registers of the Voltage L1-N (I need also Voltage L2-N, Voltage L3-N, Current L1, Current L2, Current L3, Active Power Total):

[image]

with this code:

```
@Override
protected ModbusProtocol defineModbusProtocol(int unitId) {
		
		return new ModbusProtocol(unitId, new FC3ReadRegistersTask(0x5B00, Priority.HIGH,
				cm(new UnsignedWordElement(0x5B00))
						.m(AsymmetricMeter.ChannelId.VOLTAGE_L1, ElementToChannelConverter.SCALE_FACTOR_1).build()));
	}
```
What does this error mean and why do I receive this? I also get the same error when I use your demo application (io.openems.edge.meter.socomec.dirisa14) with my modbus configuration...

Furthermore, this arises some more questions:

1. Where is the GUI for that meter? Do I have to configure it and how?
2. In the manual you've mentioned that the UI for configuring the devices will be available as soon as the migration to OSGI has been finished. Is it not finished, yet? Thus, I don't see anything when I press the settings button, except some system logs, on the right side:

[image]

3. At any new start of the .jar file I've to configure things again and again.... Is there any way to configure the felix osgi directly in java?

Here is my modbus configuration:

[image]

Here is my meter device configuration:

[image]

My device code:

```
package io.openems.edge.meter.b23.abb;

import org.osgi.service.cm.ConfigurationAdmin;
import org.osgi.service.component.ComponentContext;
import org.osgi.service.component.annotations.Activate;
import org.osgi.service.component.annotations.Component;
import org.osgi.service.component.annotations.ConfigurationPolicy;
import org.osgi.service.component.annotations.Deactivate;
import org.osgi.service.component.annotations.Reference;
import org.osgi.service.component.annotations.ReferenceCardinality;
import org.osgi.service.component.annotations.ReferencePolicy;
import org.osgi.service.component.annotations.ReferencePolicyOption;
import org.osgi.service.metatype.annotations.Designate;

import io.openems.edge.bridge.modbus.api.AbstractOpenemsModbusComponent;
import io.openems.edge.bridge.modbus.api.BridgeModbus;
import io.openems.edge.bridge.modbus.api.ElementToChannelConverter;
import io.openems.edge.bridge.modbus.api.ModbusProtocol;
import io.openems.edge.bridge.modbus.api.element.DummyRegisterElement;
import io.openems.edge.bridge.modbus.api.element.SignedDoublewordElement;
import io.openems.edge.bridge.modbus.api.element.UnsignedDoublewordElement;
import io.openems.edge.bridge.modbus.api.element.UnsignedWordElement;
import io.openems.edge.bridge.modbus.api.task.FC3ReadRegistersTask;
import io.openems.edge.common.channel.doc.Doc;
import io.openems.edge.common.component.OpenemsComponent;
import io.openems.edge.common.taskmanager.Priority;
import io.openems.edge.meter.api.AsymmetricMeter;
import io.openems.edge.meter.api.MeterType;
import io.openems.edge.meter.api.SymmetricMeter;

@Designate(ocd = Config.class, factory = true)
@Component(name = "Meter.ABB.b24", immediate = true, configurationPolicy = ConfigurationPolicy.REQUIRE)
public class MeterABBb23b24 extends AbstractOpenemsModbusComponent
		implements SymmetricMeter, AsymmetricMeter, OpenemsComponent {

	public MeterABBb23b24() {
		Utils.initializeChannels(this).forEach(channel -> this.addChannel(channel));
	}
	
	private MeterType meterType = MeterType.PRODUCTION; 

	@Reference(policy = ReferencePolicy.STATIC, policyOption = ReferencePolicyOption.GREEDY, cardinality = ReferenceCardinality.MANDATORY)
	protected void setModbus(BridgeModbus modbus) {
		super.setModbus(modbus);
	}

	@Reference
	protected ConfigurationAdmin cm;

	@Activate
	void activate(ComponentContext context, Config config) {
		this.meterType = config.type();

		super.activate(context, config.service_pid(), config.id(), config.enabled(), config.modbusUnitId(), this.cm,
				"Modbus", config.modbus_id());

		// Initialize Min/MaxActivePower channels
		this._initializeMinMaxActivePower(this.cm, config.service_pid(), config.minActivePower(),
				config.maxActivePower());
	}

	@Deactivate
	protected void deactivate() {
		super.deactivate();
	}

	public enum ChannelId implements io.openems.edge.common.channel.doc.ChannelId {
		;
		private final Doc doc;

		private ChannelId(Doc doc) {
			this.doc = doc;
		}

		public Doc doc() {
			return this.doc;
		}
	}

	@Override
	public MeterType getMeterType() {
		return this.meterType;
	}

	@Override
	protected ModbusProtocol defineModbusProtocol(int unitId) {
		
		return new ModbusProtocol(unitId, new FC3ReadRegistersTask(0x5B00, Priority.HIGH,
				cm(new UnsignedWordElement(0x5B00))
						.m(AsymmetricMeter.ChannelId.VOLTAGE_L1, ElementToChannelConverter.SCALE_FACTOR_1).build()));
	}
	@Override
	public String debugLog() {
		return "Power:" + this.getReactivePowerL1().value().asString();
	}

}
```
I'd be grateful for the support.

sfeilmeier, 2018-08-17:
Hi,

I remember seeing this error, but I am not sure where and how I solved it. Do you have a public branch with your current code which I could clone to have a look?

The target property is that long LDAP-like filter which you can see in "Modbus target filter". It is generated automatically via the call to super.activate(). I am not sure, why it could be invalid...

For the other questions:
1. Which GUI do you mean? The Energymonitor should already work with the configurations you made. All the calculations (grid, production, consumption) are made by the "_sum" component in 'io.openems.edge.core'

2. No, we (FENECON) still have most of our productive systems running on the Non-OSGi version in old_master branch. The migration is a project we are working on intensively, but it is not finished yet. A couple of our systems are already running in production on the OSGi version though. Configuration screens are going to come with this migration, too. They will essentially do the same as the Apache Felix Web interface, but will be available via the OpenEMS UI and also routed via OpenEMS Backend. This migration comes also with a lot of UX changes in the UI for easy provisioning and setting up via setup assistants. Your input is highly appreciated in any case.

3. ConfigurationAdmin uses an environment variable to set the config directory. By default this is set to 'c:/openems-config'. It can be set in the EdgeApp.bndrun file (https://github.com/OpenEMS/openems/blob/develop/io.openems.edge.application/EdgeApp.bndrun#L9)

PlotlyUser2, 2018-08-18:
No, there is no public branch of my code, and it seems to me not necessary to provide it, because I took 95% of your code and just entered my registers... ;)

1. I thought that I'd be able to configure the GUI myself, e.g. just show the voltages L1/L2/L3 in a separate card window, like home assistant does:

[image]

I've tested the Modbus TCP via this tool: https://sourceforge.net/projects/pymodslave/
and as you mentioned I'm able to see the power consumption on the main window (further, I've not seen this invalid syntax error) :

[image]

Now, is it possible to configure or customize the GUI as I want to?

I'm now playing around for at least 1 week with openEMS and if you don't mind, I'd like to give you in general a short feedback:

1. I find it very confusing, why any prospective customer have to download the source and build it in eclipse? Why does not simply provide an already built software including the server, like openHAB2 does?

2. The within code configuration of the modbus is way too complicated. Why do I have to write a whole binding to read some values from the registers? Please, take a look at home assistant:
https://www.home-assistant.io/components/sensor.modbus/
```
# Example configuration.yaml entry
sensor:
  platform: modbus
  registers:
    - name: Sensor1
      unit_of_measurement: °C
      slave: 1
      register: 100
    - name: Sensor2
      unit_of_measurement: mg
      slave: 1
      register: 110
      count: 2
```
That is a customer friendly solution and it would be much better if it can be done within GUI.

sfeilmeier, 2018-08-18:
I was just asking in order to help you better with that error. Also it would be nice if you could contribute the implementation of the ABB B24 meter to this open source project.

There is no such feature to configure/customize the UI dynamically as you described. It is also not on our (FENECON) immediate development agenda, as it is not a requirement we currently see for our customers. If you wanted to show Voltages (or any other Channel value) the way to go currently is via adding the Channels to the subscribed Channels and customizing the "Energytable" in the UI source code.

Thank you for your valuable feedback. Please keep in mind, that OpenEMS is a rather young open source project and has a very different target audience than OpenHAB and HomeAssistant. For a large part it was developed with the requirements that we (FENECON) had for our projects and products around energy storage systems. Most setups are currently preconfigured for specific standard products (i.e. an energy strorage system combined with a specific grid meter, a specific charging station,...) and the end user is not expected/required to do manual adjustments. This is also why there is currently no such simple Modbus implementation like you suggested - while of course it would be technically possible to implement. 

Having said this, we are very open for collaboration and are happy to improve OpenEMS in the directions that are required by its users. In the end this is what an open source project is all about. Also - as I asked you already in another Issue - it would be interesting to know your objective for using OpenEMS, in order to understand your specific requirements and support you.

sfeilmeier, 2018-08-18:
We see a difference between HEMS (Home Energy Management System) and HAS (Home Automation System). See for example [this presentation by Porsche AG](http://www.cpuc.ca.gov/WorkArea/DownloadAsset.aspx?id=6442453755), page 30.

From my understanding I would consider OpenHAB/Eclipse SmartHome and HomeAssistant to be a HAS in this context. OpenEMS has a stronger focus in HEMS and commercial/industrial energy management applications (see [this project](https://www.industr.com/de/stabile-stromversorgung-in-der-city-2333131) ([video presentation](https://www.youtube.com/watch?v=4MjKmsJ8On0)), which was implemented with OpenEMS)

Of course the boundaries between those two fields often overlap.

closed 2019-01-23 by sfeilmeier