OpenEMS/openems #86: Float values from registers being mishandled

KyleMicrocare, 2018-08-30:
### Bug Report or Feature Request (mark with an `x`)
```
- [ x ] bug report -> please search issues before submitting
- [ ] feature request

### Bug description or desired functionality.
Float values are being misread from the SDM630 registers, causing reported values on the UI to fluctuate wildly between massive positive and negative values:

![image](https://user-images.githubusercontent.com/42207179/44839497-3b250880-ac3f-11e8-9791-f8ee708319bb.png)

sfeilmeier, 2018-08-30:
Hi Kyle,

Thanks for reporting. Are the values correct in the system? Have you tried printing them using DetailedLogController or receiving them via [REST-Api](https://github.com/OpenEMS/openems/blob/develop/doc/OpenEMS_EN.adoc#4512-api-restjson-controller)? Do you see the correct values there?

KyleMicrocare, 2018-08-30:
I'll give that a try

KyleMicrocare, 2018-08-30:
Nope, values are still incorrect:
`2018-08-30 10:40:06,175 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] =======================================
2018-08-30 10:40:06,175 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ID                             : meter0
2018-08-30 10:40:06,176 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ---------------------------------------
2018-08-30 10:40:06,176 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ActivePower                    :     -2147483648 W   
2018-08-30 10:40:06,176 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ActivePowerL1                  :     -2147483648 W   
2018-08-30 10:40:06,176 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ApparentPower                  :               0 VA  
2018-08-30 10:40:06,176 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ApparentPowerL1                :               0 VA  
2018-08-30 10:40:06,176 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] CurrentL1                      :         5210112 mA  
2018-08-30 10:40:06,177 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ---------------------------------------
2018-08-30 10:40:07,175 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] =======================================
2018-08-30 10:40:07,175 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ID                             : meter0
2018-08-30 10:40:07,175 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ---------------------------------------
2018-08-30 10:40:07,176 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ActivePower                    :               0 W   
2018-08-30 10:40:07,176 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ActivePowerL1                  :               0 W   
2018-08-30 10:40:07,176 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] Voltage                        :               0 mV  
2018-08-30 10:40:07,176 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ---------------------------------------
2018-08-30 10:40:08,174 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] =======================================
2018-08-30 10:40:08,174 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ID                             : meter0
2018-08-30 10:40:08,174 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ---------------------------------------
2018-08-30 10:40:08,174 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ActivePower                    :      2147483647 W   
2018-08-30 10:40:08,174 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ActivePowerL1                  :      2147483647 W   
2018-08-30 10:40:08,174 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] CurrentL1                      :      2147483647 mA  
2018-08-30 10:40:08,174 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ReactivePower                  :           27424 var 
2018-08-30 10:40:08,175 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] VoltageL1                      :               0 mV  
2018-08-30 10:40:08,175 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ---------------------------------------
2018-08-30 10:40:09,174 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] =======================================
2018-08-30 10:40:09,174 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ID                             : meter0
2018-08-30 10:40:09,174 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ---------------------------------------
2018-08-30 10:40:09,175 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] CurrentL1                      :               0 mA  
2018-08-30 10:40:09,175 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ReactivePower                  :               0 var 
2018-08-30 10:40:09,175 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ---------------------------------------
2018-08-30 10:40:10,173 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] =======================================
2018-08-30 10:40:10,174 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ID                             : meter0
2018-08-30 10:40:10,174 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ---------------------------------------
2018-08-30 10:40:10,174 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ApparentPower                  :     -2147483648 VA  
2018-08-30 10:40:10,175 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ApparentPowerL1                :     -2147483648 VA  
2018-08-30 10:40:10,175 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ---------------------------------------
2018-08-30 10:40:11,173 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] =======================================
2018-08-30 10:40:11,174 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ID                             : meter0
[... cut ...]
2018-08-30 10:40:12,174 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ApparentPowerL1                :      2147483647 VA  
2018-08-30 10:40:12,174 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] CurrentL1                      :            -508 mA  
2018-08-30 10:40:12,174 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ReactivePowerL1                :               0 var 
2018-08-30 10:40:12,174 [Executor] INFO  [g.detailedlog.DebugDetailedLog] [ctrlDetailedLog0] ---------------------------------------`

KyleMicrocare, 2018-08-30:
I may be using the incorrect scale factors, but I'm not sure that would be enough to cause such a large buffer overflow error, and the values are still correct where the correct scale factors are being used. I'll try with the correct scale factors on all elements.

sfeilmeier, 2018-08-30:
Then the problem is not in the UI, but in the modbus definition. From the big numbers I would recommend validating again WordOrder and ByteOrder.

KyleMicrocare, 2018-08-30:
I didn't consider ByteOrder. I'll give that a go

KyleMicrocare, 2018-08-30:
All permutations of WordOrder and ByteOrder options produce nonsense values. I've double checked the addresses too. I'm honestly at my wit's end

sfeilmeier, 2018-08-30:
Can you please point me to the defineModbusProtocol() in your branch and provide the protocol definition (PDF, Excel, Webpage,...) so that I can have a look?

KyleMicrocare, 2018-08-30:
So as it turns out, the issue was the offset I was using on the modbus addresses. by incrementing the offset by 1, I was no longer straddling two 16-bit registers from two completely different meter values:

`//incorrect offset                                  
|L1_current<16bit><16bit>||<16bit><16bit>L2_current|
             joining this one^ and  ^this one
//correct offset
|L1_current<16bit><16bit>||<16bit><16bit>L2_current|
 joining this one^ and  ^this one
`

closed 2018-08-30 by KyleMicrocare