OpenEMS/openems #108: Update Implementing device Getting Started

abraovic, 2018-10-18:
### Bug Report or Feature Request (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
```

### Bug description or desired functionality.
Update Getting started tutorial on how to Implement a new device step 13. Resolving a new requirement never runs and it needs to be added manually.

abraovic, 2018-11-20:
After some time Eclipse crashes(OutOfMemoryException)
I tried 3 versions of the bndtool 3.5, 4,0 & 4.1, same behavior, so it looks like  „[osgi cycle dependecy problem](https://baptiste-wicht.com/posts/2010/03/osgi-and-cyclic-dependencies.html )“, to me.
 
I collected Java Heap dump after Eclipse crashes and it is clear that class `org.apache.felix.resolver.util.CandidateSelector` is „eating“ memory.
Class has one List which explodes in our case, check code on [link](https://github.com/eclipse/rt.equinox.framework/blob/master/bundles/org.eclipse.osgi/felix/src/org/apache/felix/resolver/util/CandidateSelector.java)

Then I did exercise by manually adding/removing bundles from the project.
So, when bundles:
```
io.openems.edge.ess.kaco.blueplanet.gridsave50          
io.openems.edge.ess.streetscooter    
```    
 
are removed then Resolve does not crash the Eclipse, and looks starting of the  OpenEMS looks OK to me.
 
Now if you new bundle like, `io.openems.edge.meter.socomec.dirisa14`, is added in Run Req., Resolve works fine.

sfeilmeier, 2018-11-29:
Thanks for reporting this problem and digging into the solution. "Resolve" using bndtools/Eclipse IDE should be fixed with commit https://github.com/OpenEMS/openems/commit/d39060f744a4a4929f1966e0328131108d919401. The docs in https://openems.io/openems/latest/edge/implement.html had been changed anyway a while ago.

closed 2018-11-29 by sfeilmeier