OpenEMS/openems #353: Fail to run getting started insructions

tomaszG247, 2019-02-27:
### Bug Report or Feature Request (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
```

### Bug description or desired functionality.
I can't make the basic install work. I followed the  installation instructions, except that I laredy had Eclipse installed
I am using : 
-Eclipse IDE for Java Developers, Version: 2018-09 (4.9.0)
- Java 1.8.0_021
Bndtools 4.1.0-20181018

At the step : [run OSGI] for EdgeApp.bndrun

I get the not so explicit error

An internal error occurred during: "Launching EdgeApp.bndrun".
java.lang.NullPointerException

Any troubleshooting advice ?

Best

sfeilmeier, 2019-03-02:
Hi Tomasz,

thanks for getting in touch with us. That error sounds indeed weird. To give some more clues:
- Do you have any message in "Problems"? Like build errors,...
- Is "Resolve" working in EdgeApp.bndrun?
- Are you able to run BackendApp.bndrun?

By the way: I just setup a Discourse instance for OpenEMS at https://community.openems.io. Hopefully this will help us build a community of developers to help each other faster. Feel free to contact me/us there or here on GitHub. 

Regards,
Stefan

tomaszG247, 2019-03-07:
- No problems during build (I suppose they show up in Eclipse's problems pane ?)
- Resolve seems to work. 
- Backend.bndrun seems to work also.

I guess I would to do a minimal test of my bndtools install but since I am quite unfamiliar with this framework, I need to figure out how to do that.

Best

sfeilmeier, 2019-03-07:
So that is indeed weird. Maybe the best is to start again with "Getting Started" from scratch and setting up everything again. If it happens again, please post a Screenshot - maybe that will give some clue.

tomaszG247, 2019-03-07:
Hi can you please tell me which versions of BndTools you use. I found out that 4.1 seems recent and the basic tutorial on BndTools website is not up to date so I cant even test a basic project.

sfeilmeier, 2019-03-07:
I also use BndTools 4.1.0 REL.

If you are interested I could try to setup a simple workspace for you via Teamviewer. Just to find out, what might be wrong on your side.

closed 2019-07-07 by sfeilmeier