OpenEMS/openems #489: Apache Felix ArrayIndexOutOfBoundsException

Tobii42, 2019-04-15:
### Bug Report or Feature Request (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
```

### Bug description or desired functionality.
Hi,

I'm trying out OpenEMS (very interesting project!) and going through the Getting Started Guide. The point where I have to activate "Scheduler All Alphabetically" I always receive an error in the Web Console:

**AJAX Error**
```
The request failed:
HTTP ERROR: 500
Problem accessing /system/console/configMgr/[Temporary%20PID%20replaced%20by%20real%20PID%20upon%20save]. Reason:

    java.lang.ArrayIndexOutOfBoundsException: 0
```

In the console output in Eclipse, the following stack trace is printed:
```

[qtp1916904573-27] DEBUG org.eclipse.jetty.server.HttpInput - HttpInputOverHTTP@71ae0c01[c=230,s=STREAM] read 1 from Content@2251a4ab{HeapByteBufferR@5d61dabe[p=810,l=811,c=8192,r=1]={POST /system/cons...Ccontrollers.id<<<s>>>\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00...\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00}}
[qtp1916904573-27] DEBUG org.eclipse.jetty.server.HttpConnection - releaseRequestBuffer HttpConnection@770b7b1f[SelectChannelEndPoint@1767f2a7{/127.0.0.1:51876<->8080,Open,in,out,-,-,15/60000,HttpConnection}{io=0/0,kio=0,kro=1}][p=HttpParser{s=END,231 of 231},g=HttpGenerator@5fd14d93{s=START},c=HttpChannelOverHttp@3da11bac{r=1,c=false,a=DISPATCHED,uri=//localhost:8080/system/console/configMgr/[Temporary%20PID%20replaced%20by%20real%20PID%20upon%20save]}]
[qtp1916904573-27] DEBUG org.eclipse.jetty.server.HttpInput - HttpInputOverHTTP@71ae0c01[c=231,s=STREAM] consumed Content@2251a4ab{HeapByteBufferR@5d61dabe[p=811,l=811,c=8192,r=0]={POST /system/cons...controllers.ids<<<>>>\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00...\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00}}
[qtp1916904573-27] DEBUG org.eclipse.jetty.server.HttpInput - HttpInputOverHTTP@71ae0c01[c=231,s=STREAM] consumed EOF
[qtp1916904573-27] ERROR org.apache.felix.http.jetty - Exception while processing request to /system/console/configMgr/[Temporary PID replaced by real PID upon save]
java.lang.ArrayIndexOutOfBoundsException: 0
	at org.apache.felix.cm.impl.ConfigurationManager.createConfiguration(ConfigurationManager.java:964)
	at org.apache.felix.cm.impl.ConfigurationManager.createFactoryConfiguration(ConfigurationManager.java:496)
	at org.apache.felix.cm.impl.ConfigurationAdminImpl.createFactoryConfiguration(ConfigurationAdminImpl.java:101)
	at org.apache.felix.webconsole.internal.configuration.ConfigAdminSupport.getConfiguration(ConfigAdminSupport.java:150)
	at org.apache.felix.webconsole.internal.configuration.ConfigAdminSupport.applyConfiguration(ConfigAdminSupport.java:191)
	at org.apache.felix.webconsole.internal.configuration.ConfigManager.doPost(ConfigManager.java:160)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:644)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:725)
	at org.apache.felix.webconsole.internal.servlet.OsgiManager.service(OsgiManager.java:567)
	at org.apache.felix.webconsole.internal.servlet.OsgiManager$3.run(OsgiManager.java:465)
	at java.security.AccessController.doPrivileged(Native Method)
	at org.apache.felix.webconsole.internal.servlet.OsgiManager.service(OsgiManager.java:461)
	at org.apache.felix.http.base.internal.handler.ServletHandler.handle(ServletHandler.java:85)
	at org.apache.felix.http.base.internal.dispatch.InvocationChain.doFilter(InvocationChain.java:79)
	at org.apache.felix.http.base.internal.dispatch.Dispatcher.dispatch(Dispatcher.java:124)
	at org.apache.felix.http.base.internal.DispatcherServlet.service(DispatcherServlet.java:61)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:725)
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:845)
	at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:583)
	at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:224)
	at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1160)
	at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:511)
	at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:185)
	at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1092)
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:141)
	at org.eclipse.jetty.server.handler.ContextHandlerCollection.handle(ContextHandlerCollection.java:213)
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:134)
	at org.eclipse.jetty.server.Server.handle(Server.java:518)
	at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:308)
	at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:244)
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:273)
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:95)
	at org.eclipse.jetty.io.SelectChannelEndPoint$2.run(SelectChannelEndPoint.java:93)
	at org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.produceAndRun(ExecuteProduceConsume.java:246)
	at org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.run(ExecuteProduceConsume.java:156)
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:654)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$3.run(QueuedThreadPool.java:572)
	at java.lang.Thread.run(Thread.java:748)
[qtp1916904573-27] DEBUG org.eclipse.jetty.server.HttpChannel - Commit HTTP/1.1{s=500,h=4} to HttpChannelOverHttp@3da11bac{r=1,c=true,a=DISPATCHED,uri=//localhost:8080/system/console/configMgr/[Temporary%20PID%20replaced%20by%20real%20PID%20upon%20save]}
[qtp1916904573-27] DEBUG org.eclipse.jetty.server.HttpConnection - org.eclipse.jetty.server.HttpConnection$SendCallback@5d162d75[PROCESSING][i=HTTP/1.1{s=500,h=4},cb=org.eclipse.jetty.server.HttpChannel$CommitCallback@e71b843] generate: NEED_HEADER (null,[p=0,l=356,c=2048,r=356],true)@START
[qtp1916904573-27] DEBUG org.eclipse.jetty.server.HttpConnection - org.eclipse.jetty.server.HttpConnection$SendCallback@5d162d75[PROCESSING][i=HTTP/1.1{s=500,h=4},cb=org.eclipse.jetty.server.HttpChannel$CommitCallback@e71b843] generate: FLUSH ([p=0,l=181,c=16384,r=181],[p=0,l=356,c=2048,r=356],true)@COMPLETING
[qtp1916904573-27] DEBUG org.eclipse.jetty.io.WriteFlusher - write: WriteFlusher@10370767{IDLE} [HeapByteBuffer@7519e019[p=0,l=181,c=16384,r=181]={<<<HTTP/1.1 500 Serv...Length: 356\r\n\r\n>>>lhost:8080/system...\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00},HeapByteBuffer@fbe6388[p=0,l=356,c=2048,r=356]={<<<<html>\n<head>\n<me.../body>\n</html>\n>>>\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00...\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00}]
[qtp1916904573-27] DEBUG org.eclipse.jetty.io.WriteFlusher - update WriteFlusher@10370767{WRITING}:IDLE-->WRITING
[qtp1916904573-27] DEBUG org.eclipse.jetty.io.ChannelEndPoint - flushed 537 SelectChannelEndPoint@1767f2a7{/127.0.0.1:51876<->8080,Open,in,out,-,W,18/60000,HttpConnection}{io=0/0,kio=0,kro=1}
[qtp1916904573-27] DEBUG org.eclipse.jetty.io.WriteFlusher - Flushed=true 181/181+1 WriteFlusher@10370767{WRITING}
[qtp1916904573-27] DEBUG org.eclipse.jetty.io.WriteFlusher - update WriteFlusher@10370767{IDLE}:WRITING-->IDLE
[qtp1916904573-27] DEBUG org.eclipse.jetty.server.HttpConnection - org.eclipse.jetty.server.HttpConnection$SendCallback@5d162d75[PROCESSING][i=HTTP/1.1{s=500,h=4},cb=org.eclipse.jetty.server.HttpChannel$CommitCallback@e71b843] generate: DONE ([p=181,l=181,c=16384,r=0],[p=356,l=356,c=2048,r=0],true)@END
[qtp1916904573-27] DEBUG org.eclipse.jetty.server.Server - RESPONSE for /system/console/configMgr/[Temporary PID replaced by real PID upon save] h=true
500 null
Date: Mon, 15 Apr 2019 07:28:20 GMT
Cache-Control: must-revalidate,no-cache,no-store
Content-Type: text/html;charset=iso-8859-1
Content-Length: 356
```

This also happens with any other configuration that I want to apply. Is this a bug or am I doing anything wrong?

System Information:
System: Ubuntu 18.10
Java: 
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-8u191-b12-2ubuntu0.18.10.1-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
Eclipse: Version: 2019-03 (4.11.0)
felix.cm.dir in EdgeApp: /etc/openems/
OSGi Framwork in EdgeApp: org.eclipse.osgi;version='[3.10.100.v20150529-1857,3.10.100.v20150529-1857]'
Execution Env. in EdgeApp: JavaSE-1.8

Tobii42, 2019-04-15:
Sorry, nevermind, did the whole process again, now it works.

closed 2019-04-15 by Tobii42

sfeilmeier, 2019-04-18:
Hi @Tobii42,

still thank you for getting in touch and reporting the issue. I have too seen this error, but it seems to be an issue with Apache Felix. Also the stack trace shows only packages in their namespace. I was never able to figure out what was the problem - but everything seemed to work anyway.

If you have any further questions, please do not hesitate to ask here or at the community forum on http://community.openems.io.

Regards,
Stefan

Tobii42, 2019-04-23:
Hi,

I also thought that it is more an Felix issue, but I hoped that someone here maybe had the same issue. I don't know the exact solution, but for the record and maybe others with the same problem: I had this issue again and it helped to use a different directory for felix.cm.dir in EdgeApp where the EdgeApp had the permission to read and write.

Regards,
Tobias

sfeilmeier, 2019-04-23:
Ah ok. That is a very good point. I just set that path statically, but it can surely go wrong if you don't have permissions for that folder:
```
felix.cm.dir=c:/openems-config
```

https://github.com/OpenEMS/openems/blob/develop/io.openems.edge.application/EdgeApp.bndrun#L9

Thanks for finding that!