OpenEMS/openems #575: OpenEMS UI does not work, when doing a clean installation (Getting Started Guide)

lnaitsirhc, 2019-06-04:
- [x] bug report
- [ ] feature request

To reproduce the error, I

- installed openEMS exactly like described in the getting started guide.
- finally started the OpenEMS UI with `ng serve` and got the following WARNING:

  

> npm package ng2-charts throws an error  ..."export 'ɵɵdefineInjectable' was not found in '@angular/core'

 - Afterwards the website at http://localhost:4200 was not loading appropriately. I got an empty website

I was able to fix that issue, by stopping the server and installing the previous version of `ng2-charts`:
   `npm install ng2-charts@2.2.3 --save`
That fixed the problem and the UI runs fine.

 
 
**Below is a more generic solution (also tested on my system):** 
To fix the problem I would suggest, setting a fixed version for `ng2-charts` in `package.json`:
Change `"ng2-charts": "^2.2.4"`, to `"ng2-charts": "2.2.3"`,

You may find some more background information regarding the ng2-charts problem here:
[Stackoverflow](https://stackoverflow.com/questions/56371427/warning-in-node-modules-ng2-charts-fesm5-ng2-charts-js-23054-72-export-%C9%B5%C9%B5de)

sfeilmeier, 2019-06-05:
Thank you very much for this bug report and - even more - for providing a working solution! 

Do you need any other guidance or support in getting started with OpenEMS? Do you have a concrete project in mind?

@fabianfnc: please incorporate this into our packages.json; and see if it is still required to pin the version of ng2-charts with Angular 8, after we updated.

sfeilmeier, 2019-06-05:
@fabianfnc: the proposed solution in commit https://github.com/OpenEMS/openems/commit/20c5927aee7ed9c447e9b24d84b042b570ef92b2 does not solve the issue. Also there was a new major update of ng2-charts. I am not sure if it maybe already solves the issue.

fabianfnc, 2019-06-06:
The issue is still there with version 2.2.3 which was released 5 days ago.
commit https://github.com/OpenEMS/openems/commit/dfcd0d79ec9ac88dac4ac0842a469445fc096580 fixed it for now.

commit 20c5927a (2019-06-05): downgraded ng2-charts to 2.2.3 this fixed #575
closed 2019-06-06 by fabianfnc