OpenEMS/openems #614: EVCS included in consumption

Facj, 2019-06-21:
### Bug Report or Feature Request (mark with an `x`)
```
- [ ] bug report 
- [x ] feature request
```

### Bug description or desired functionality.
In order to monitor the consumption as a whole in my simulations, I would like the EVCS to be included in the Consumption.
Please let me know if it is already possible.

sfeilmeier, 2019-07-07:
@sebastianasen: I just saw that you did not reply on this topic till now. Can you please take over?

sfeilmeier, 2020-07-12:
At least in the UI the consumption of EVCS is listed separately.
https://github.com/OpenEMS/openems/blob/d084020fa78189ee117f37a6148d0156a75d19c6/ui/src/app/edge/live/consumption/consumption.component.ts#L37

We are working on a more generic 'power meter' nature that allows to solve this problem using better abstraction. I am still closing this issue finally, as the requested feature is there already.

closed 2020-07-12 by sfeilmeier