OpenEMS/openems #663: Deactivating an EDGE Bundle during RUN is not reflected in the UI and in the backend

clehne, 2019-07-19:
### Bug Report or Feature Request (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
```

### Bug description or desired functionality.
### Steps to reproduce the behaviour:
1. Configured OpenEMS Edge to work with OpenEMS Backend.
2. Used the OSGI Bundle "Simulator DataSource: CSV Reader" with Standard Weekday Load Profile
3. Added and enabled the OSGI Bundle "Simulator PV-Inverter (meter1)" Service and used "datasource0" for data input.
4. Opened OpenEMS UI in browser and watched it for a few minutes. Now we see solar generated power (due to new "meter1"-component). UI shows this fine.
5. Deactivated the OSGI Bundle "Simulator PV-Inverter (meter1)" Service  (it also doesn't make a difference if the service is deleted)
6. The OpenEMS UI in browser still holds and displays the last solar generator power. Also the backend database holds the last solar generator power (meter1)
Note that the Edge bundle seems to work properly, because in the logs "meter1" is not shown anymore 

### Is the behaviour reproducable
Yes

### Is it a critical bug
I do not think so, because as far as I know in production environment a service will typically not be disabled during one run. 

### Reason for the problem
It seems the problem comes from the "io.openems.edge.controller.api.backend" bundle.
The problem is located in `BackendWorker.java`:
1) The method `updateData()` does filter the components which are not enabled. The problem is that it is working on a hashmap. So it overwrites all entries which components are enabled. But the hashmap still also holds old values of disabled components.
2) The method `forever()` transmits all values to the backend. Most of the time it sends changed values only, so there is no way for the backend to recognise, that the component was disabled and the value needs to be removed. But even if it would send all values to the backend, the backend won't recognize that there is a deactivated component.
 
 
### Possible Solution
I am an OpeneEMS newby, but maybe the state of the hashmap could be resetted, everytime when a component is deactivated. The internal state of the backend could be resetted also.

clehne, 2019-07-22:
I played around and found a possible solution for it. The following patch installs an OSGI-ServiceListener and clears the hashmap whenever a service is changing. With this solution the backend and the UI reflects the service deactivation change immediately. I can't see any side effects with that. 
If it looks good to you also, I can open a pull request with the given solution.

Attached files:
[BackendApi.diff.txt](https://github.com/OpenEMS/openems/files/3416100/BackendApi.diff.txt)
[BackendWorker.diff.txt](https://github.com/OpenEMS/openems/files/3416102/BackendWorker.diff.txt)

clehne, 2019-07-22:
Due to hot swapping while debugging I missed an important point. The module also needs to send all values after service has changed (not only the changed ones). With the new BackednWorker everythings works as expected.

[BackendWorker.diff.txt](https://github.com/OpenEMS/openems/files/3417221/BackendWorker.diff.txt)

sfeilmeier, 2024-03-03:
We are moving all discussions to the OpenEMS Community (https://community.openems.io/), to have them in one central place. Please feel free to open a thread there.

closed 2024-03-03 by sfeilmeier