OpenEMS/openems #837: Day Ahead Scheduling Algorithm

saadsaeed123, 2019-09-30:
- [ ] bug report -> please search issues before submitting
- [x] feature request

### Bug description or desired functionality.

sfeilmeier, 2019-09-30:
Hi,

yes, this is a topic we are actively working on. Please see this thread in the community forum: https://community.openems.io/t/integrate-weather-data-for-forecasts/123

@venu-sagar is currently testing an algorithm that uses these forecasts to limit the battery charging during the day. This is not in 'develop' branch yet. He also implemented a controller that uses day ahead dynamic prices by Awattar to optimize charge/discharge behaviour of a battery during the night.

saadsaeed123, 2019-09-30:
So what is this predictor API? 
Is this API mentioned in the document somewhere?

saadsaeed123, 2019-09-30:
If I am not wrong 'same-as-the-last-day' algorithm is `simply` using the energy values of the current day as the predicted values of the next day?

sfeilmeier, 2019-09-30:
Yes, that's true. It's just meant to be a simple "dummy" implementation. Together with University of Passau we are currently developing more advanced implementations of the predictor APIs (https://openems.github.io/openems.io/openems/latest/randd.html#_emsig).

What is your use-case?

saadsaeed123, 2019-10-01:
Actually I was looking for an implementation of MILP which is widely used in DC Micro-grids as an optimization algorithm which uses day ahead prediction . It minimizes 'cost' based upon some constraints e.g. 'Grid Injection Limits', 'Storage Limits', e.t.c. Following are the papers in which these algorithm is used.
https://www.researchgate.net/publication/260913618_DC_microgrid_power_flow_optimization_by_multi-layer_supervision_control_Design_and_experimental_validation
https://ieeexplore.ieee.org/document/7514024
https://ieeexplore.ieee.org/document/7048778

sfeilmeier, 2019-12-16:
The algorithms sound indeed interesting. Unfortunately I do not have access to researchgate. Would you mind sending that pdf so I can have a look? (stefan.feilmeier at fenecon.de)

zoernert, 2021-02-15:
Day ahead scheduling might be sourced via http://api.corrently.io/v2.0/marketdata (documentation: https://corrently.io/#operation/gsiMarketdata ) - However I think this issue might be moved from "Open" to "Implemented" as most of the code exists.

closed 2024-03-03 by sfeilmeier