OpenEMS/openems #1111: missing check if nature is defined

ahummelsberger, 2020-04-17:
https://github.com/OpenEMS/openems/blob/03f2fd5445dd5f4cface94647b75fb9e147ed0a8/ui/src/app/shared/edge/edgeconfig.ts#L127-L136

Like in other functions there should be a check if nature is defined before going into the for loop.

sfeilmeier, 2020-04-20:
Hi Andreas,

yes you are right. It should never happen to be null, but a check would certainly be reasonable. Could you add it?

ahummelsberger, 2020-04-20:
> Hi Andreas,
> 
> yes you are right. It should never happen to be null, but a check would certainly be reasonable. Could you add it?

I had a problem viewing the add/install component and the profile view.
When I traced down the error, I recognized that the error was thrown because I hadn't installed any battery modules in my edge deployment. But the UI still tried to iterate through every nature, even if it wasn't available in the first place.

I will make a pull request then.

sfeilmeier, 2020-05-06:
Fixed in #1113

closed 2020-05-06 by sfeilmeier