OpenEMS/openems #1200: EventAdmin: Blacklisting ServiceReference due to timeout (Controller Api Backend)

ahummelsberger, 2020-08-12:
### Bug Report or Feature Request (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
```

### Bug description or desired functionality.

I got an issue that my edge device stops sending live data via the Controller.Api.Backend Component.
In the logs I got the message:

`WARN  [org.apache.felix.eventadmin   ] EventAdmin: Blacklisting ServiceReference [{io.openems.edge.controller.api.Controller, io.openems.edge.common.component.OpenemsComponent, org.ops4j.pax.logging.spi.PaxAppender, org.osgi.service.event.EventHandler}={..., component.name=Controller.Api.Backend, component.id=9, enabled=true, webconsole.configurationFactory.nameHint=Controller Api Backend [{id}], alias=, debug=true, event.topics=[io/openems/edge/cycle/AFTER_PROCESS_IMAGE,io/openems/edge/config/UPDATE], proxyPort=0, org.ops4j.pax.logging.appender.name=Controller.Api.Backend, service.factoryPid=Controller.Api.Backend, apiTimeout=60, id=ctrlBackend0, service.pid=Controller.Api.Backend.93b94005-37a1-4246-9dc9-6c7cfc0c93a6, proxyType=HTTP} | Bundle(io.openems.edge.controller.api.backend_1.0.0.202008051004 [16])] due to timeout!`

I couldn't find out yet where the problem is, but I guess it has to be somewhere in the BackendWorker Class because anything else works fine like showing the debug logs on the backend side, config updates, history...

I was able to bypass this bug by adding `-Dorg.apache.felix.eventadmin.Timeout=0` to the launch of the Edge application.
But usually you should not do that because the problem should be solved inside the code of the event handling.

sfeilmeier, 2020-08-18:
Hi Andreas,

I don't think the problem can be in the BackendWorker because that runs entirely in a separate thread. But the call to `triggerNextRun()` is really the only call in the event handler... (https://github.com/OpenEMS/openems/blob/develop/io.openems.edge.controller.api.backend/src/io/openems/edge/controller/api/backend/BackendApi.java#L198) ... very strange.

As you wrote already, I activated the felix timeout globally by default (https://github.com/OpenEMS/openems/blob/develop/io.openems.edge.application/EdgeApp.bndrun#L11), because in my opinion it can really lead to unwanted situations. 

Honestly I do not have any idea, why this happened to you. Can you provide more debug information or try to trace the problem? Otherwise I believe I have to close the issue as 'non reproducible'.

ahummelsberger, 2020-08-20:
Hi Stefan,

I hadn't much time to trace the problem yet. You can close this issue as the felix timeout works for me at the moment.
When I'm able to trace it down, I'll let you know.

closed 2020-08-20 by sfeilmeier