OpenEMS/openems #1226: Discovergy Meter: JSON parsing error parsing for 'power1'

martingruening, 2020-09-12:
### Bug Report or Feature Request (mark with an `x`)
```
- [X] bug report -> please search issues before submitting
- [ ] feature request
```

### Bug description or desired functionality.

Added a Discovergy smart meter as 'disco0' (Two way meter, but currently only consumption from grid direction is used)
Debug log on edge shows:
```
INFO  [e.controller.debuglog.DebugLog] [ctrlDebugLog0] disco0[L:548 W|State:FAULT: RestApiFailed]
[disco0  ] ERROR [s.common.worker.AbstractWorker] Worker error. OpenemsNamedException: JSON [power1] is not a member of [{"energyOut":1802000000,"energy2":0,"energy1":38409877000000,"power":707800,"energyOut1":18020000...]
io.openems.common.exceptions.OpenemsError$OpenemsNamedException: JSON [power1] is not a member of [{"energyOut":1802000000,"energy2":0,"energy1":38409877000000,"power":707800,"energyOut1":18020000...]
	at io.openems.common.exceptions.OpenemsError.exception(OpenemsError.java:149)
	at io.openems.common.utils.JsonUtils.getSubElement(JsonUtils.java:648)
	at io.openems.common.utils.JsonUtils.getAsPrimitive(JsonUtils.java:554)
	at io.openems.common.utils.JsonUtils.getAsInt(JsonUtils.java:107)
	at io.openems.edge.meter.discovergy.DiscovergyWorker.forever(DiscovergyWorker.java:82)
	at io.openems.common.worker.AbstractWorker$1.run(AbstractWorker.java:112)
```

The item 'power1' seems not to be part of the JSON meter reading provided by the Discovergy API. Potentially this is because only the consumption from grid direction is used.

For reference this is a sample full 'value' part of my meter's readings:
`"values":{"energyOut":1802000000,"energy2":0,"energy1":38414927000000,"energyOut1":1802000000,"power":3005500,"energyOut2":0,"energy":38414927000000}`

sfeilmeier, 2020-09-15:
Thanks Martin for reporting this issue and contributing to OpenEMS!

I adjusted the code so that it ignores non-available fields in the JSON. Could you please test if #1230 fixes the issue for you?

You can test the bugfix also with the newly implemented Gitpod integration: https://gitpod.io/#https://github.com/OpenEMS/openems/tree/feature/fix-discovergy-1226 

(see a770d4ab19310f0be5477222464ce6e625c18e15 for details on Gitpod)

closed 2020-09-30 by sfeilmeier

martingruening, 2020-10-19:
I took me some time to build a new version of the edge component. But I can confirm that the issue is fixed. Thanks!

sfeilmeier, 2020-10-19:
Thanks for the feedback!