OpenEMS/openems #1233: UI: no reaction on grid mode changes

ahummelsberger, 2020-09-18:
### Bug Report or Feature Request (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
```

### Bug description or desired functionality.

I tested the ui behavior when my test system changes to off grid and found out that the channel _sum/GridMode was never subscribed by the UI.
So the grid icon on the energymonitor never changed and also no change in the Grid widget.
I made a bugfix by just adding 
`new ChannelAddress('_sum', 'GridMode')`
in energymonitor.component.ts

Shall I do a pull request for that?

sfeilmeier, 2020-09-18:
@fabianfnc: I remember we used to have this in the Energy Monitor? Can you see what happend to this?

fabianfnc, 2020-09-21:
@ahummelsberger I will take care of this. Thanks for the Info!

closed 2020-09-30 by sebastianasen

sfeilmeier, 2020-09-30:
@ahummelsberger: Can you please validate if the new behaviour works for you?

ahummelsberger, 2020-10-01:
@sfeilmeier yes it does, thank you!