OpenEMS/openems #1346: OpenEMS in ioBroker einbinden

Stefan231278, 2020-12-28:
Hallo,

ich möchte meine FEMS im ioBroker implementieren und z.B. die Daten der Batterie in die Scripte mit einbauen. Ich kann zwar über den Modbus eine Verbindung herstellen, bekomme aber keine Daten zurück.

Wie klappts am besten?

Danke für eine Antwort.

martingruening, 2020-12-29:
Ich würde die REST API von openems-edge auf Port 8084 verwenden, um die gewünschten Daten abzufragen. Das ist extrem einfach und funktioniert z.B. mit openHAB einwandfrei. Das sollte auch mit ioBroker kein Problem sein.

Stefan231278, 2020-12-29:
Danke, aber ich bekomme da immer die Meldung:

{"jsonrpc":"2.0","id":"00000000-0000-0000-0000-000000000000","error":{"code":1003,"message":"Authentication failed","data":[]}}

Obwohl ich doch normalerweise mit dieser Adresse http://x:user@192.168.178.43:8084/rest darauf zugreifen müssen könnte.
Welcher Username und Passwort ist denn richtig. Lt. Dokumentation des FEMS gibt es keinen Username, und das Passwort ist user (wenn man im Heimnetz ist)

LG
Stefan

martingruening, 2020-12-30:
Hallo Stefan,

github ist kein Diskussionsforum. Deswegen schlage ich vor, dass du deine konkrete Fragestellung in der OpenEMS Community einbringst ([https://community.openems.io/](https://community.openems.io/)). Dort helfe ich dir gerne weiter.

closed 2023-03-28 by hydroid7