OpenEMS/openems #1347: [API] Auth error in HTTP request to API should trigger a 401 response code (not 400)

martingruening, 2020-12-29:
If the authentication fails during an HTTP request to the API it should return a 401 (Unauthorized)
[https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/401](https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/401)

Why is that important?
First: it is the correct response for this situation.

Second: the popular jetty HTTP client library will first issue a request WITHOUT credentials (even when configured with credentials) and only if it receives a 401 response will redo the request with credentials. That means as is you can't query the API with a jetty client (like used in the OpenHAB http binding). The reason is then quite hard to find (took me some hours).

martingruening, 2021-01-02:
Already fixed, this can be assigned to me. I will submit a PR for RestHandler that includes this change.

sfeilmeier, 2021-01-02:
Thank you! Right now I do have some other priorities, but will be back on track soon.

closed 2021-04-09 by martingruening