OpenEMS/openems #1348: [API] Insert Unit into channel output

martingruening, 2020-12-30:
Feature Request:
When reading a channel through the REST API also provide the unit of the channel.

Example - instead of
```
{
    "value": 457,
    "type": "integer",
    "writable": false
}
```

send
```
{
    "value": 457,
    "unit": "W",
    "type": "integer",
    "writable": false
}
```

martingruening, 2020-12-31:
You can assign this to me. Done the easy modfication already. PR will follow bundled with other improvements in RestHandler.

sfeilmeier, 2021-01-02:
A quick thought on this: if I would implement it, I would try to reuse the Channel representation in `EdgeConfig`. EdgeConfig is generally used to represent the entire configuration of an OpenEMS Edge and is used in OpenEMS UI to build the interface. See:

https://github.com/OpenEMS/openems/blob/develop/io.openems.common/src/io/openems/common/types/EdgeConfig.java#L236

martingruening, 2021-01-02:
Good suggestion. Will do that. Shall I keep the existing "writable" property in the API code? This would make sure that we don't break code depending on the API, but then the writable property seems to be redundant with the "accessMode" property of the config.

sfeilmeier, 2021-01-02:
I believe dropping the "writable" property is ok.

closed 2021-04-09 by martingruening