OpenEMS/openems #1382: Error while deploying OpenEms Edge on an Odriod (Ubuntu 20.0.4)

bilalh96, 2021-02-01:
Hi Team Openems

[image]
 I am facing this error while deploying OpenEms edge on my Odroid N2. Can you please tell me whats the cause of this error? I am not running the code on eclipse IDE at the moment on my computer. as I am new to this platform and also the other applications including eclipse, java language, so I am not able to set it on my device correctly. can you please give an easy-to-understand guide on how to fix this error. Yes, I have followed the steps mentioned in the Getting started guide already.
 
 Yours sincerely
 Bilal

### Bug description or desired functionality.

sfeilmeier, 2021-02-01:
Hi Bilal,

in general OpenEMS Community forum is the better place to discuss these problems, and I recommend it for you to get in touch with other community members (-> https://community.openems.io/).

After you edit a systemd service via `sudo nano /etc/systemd/system/openems.service` you will have to tell systemd about it. Use the command `systemctl daemon-reload` to do that.

Then try again if you can start the OpenEMS service just like you did, using `sudo systemctl restart openems`. `journalctl -lfu openems` then shows the log outputs. 

You can also bypass the systemd service for now and just run OpenEMS via `/usr/bin/java -Dfelix.cm.dir=/etc/openems.d/ -jar /usr/lib/openems/openems.jar`. Just be aware that it will not start again then, once you reboot your Odroid board.

Regards,
Stefan

bilalh96, 2021-02-01:
Hi Stefan,

I followed the exact steps you mentioned above but I get the same error :(  

Best regards,
Bilal

sfeilmeier, 2021-02-01:
Can you please copy&paste the log when you execute manually using `/usr/bin/java -Dfelix.cm.dir=/etc/openems.d/ -jar /usr/lib/openems/openems.jar`?

bilalh96, 2021-02-01:
[image]
[image]

ok now its visible but why am I getting no schedulers configured error ? do I need to run the java openems edge run file from eclipse IDE everytime or how can I set the schedulers ? before like I opened them for eg the Apache web config webpage and saw the settings changes in the eclipse logs , plus for the GUI as mentioned I was using angular in visual code studio to start ng serve etc . how can all of this implemented now in the case where I have this jar file on my odroid only?

sfeilmeier, 2021-02-01:
The parameter `-Dfelix.cm.dir=/etc/openems.d/` tells OpenEMS (or better the Apache Felix Configuration Admin), that it should use that directory for the configuration files. As you run OpenEMS for the first time on this device, this directory will be mostly empty. This is the same situation as in the Getting Started Guide, before you start configuring in chapter 3 (https://openems.github.io/openems.io/openems/latest/gettingstarted.html#_run_openems_edge_and_start_simulator)

Now you can either:
- Open Apache Felix on your Odroid via http://ODROID-IP:8080/system/console/configMgr and follow the Getting Started Guide
- Copy the configuration from your laptop (if you are on windows, it will most likely be stored in `c:/openems/config`, which is a preset value defined [here](https://github.com/OpenEMS/openems/blob/develop/io.openems.edge.application/EdgeApp.bndrun)) to your Odroid Linux e.g. via WinSCP
- Use the OpenEMS example configuration from https://github.com/OpenEMS/openems/tree/develop/tools/docker/openems-edge/config.d (which we also use in our [Gitpod demo](https://gitpod.io/#https://github.com/OpenEMS/openems))

bilalh96, 2021-02-01:
Hi Stefan

So first method didnt worked for me , the webpage never loaded, for the second one I copied the conf folder from c:/openems/config to the folder to the directory where the openems.jar file is located  and then run the 

this time when I ran Execute systemctl restart openems --no-block; journalctl -lfu openems , it does open like it was in the deployment guide but as you mentioned earlier with sudo systemctl restart openems. journalctl -lfu openems , this dosent run command gives errror on "-u , command not found". 
However even after copying the config folder , I see the same errors like " there are no schedulers configured" , " Uable to start modbus/tcp api on port [502] : cannot start listner " etc .. 
I am stuck in the same issue :( and also how can I see the GUI like I was seeing on my windows.

Yours sincerely
Bilal

sfeilmeier, 2021-02-01:
Ok, I believe you are unfortunately missing some basic required knowledge here. Also the part of deployment on IoT hardware is not very well documented yet, unfortunately. Could we continue the discussion in the community forum, so that future users can also benefit from your findings? Could you please open a thread there?

Regarding setting up the UI, there is a thread here: https://community.openems.io/t/ui-am-raspberry-konfigurieren/72 (its in German, but you should be able to translate it).

Thanks,
Stefan

bilalh96, 2021-02-01:
done thanks ,  I have opened a  thread there , you can delete this issue thread now.

bilalh96, 2021-02-02:
Hi Stefan ,

Could you please reply to the new thread which I have opened in the
community forum. I am waiting your reply there.

Thank you so much

Yours sincerely
Bilal

sfeilmeier, 2021-02-02:
Continuing in the OpenEMS Community Forum: https://community.openems.io/t/error-while-deploying-openems-edge-on-an-odriod-ubuntu-20-0-4/414/2

closed 2021-02-02 by sfeilmeier