OpenEMS/openems #1385: Problem with displaying historical data

zoesn89, 2021-02-04:
### Bug Report or Feature Request (mark with an `x`)
```
- [x] bug report
- [ ] feature request
```

### Bug description
Hey,

I recently installed openems edge and I'm testing out the features. When I'm on the page with the historical data, I have some graphical problems (screenshot). I'm also wondering, why my overall consumption and production data isn't displayed. Do I need an extra controller for that?

Here is a [screenshot](https://s1.imagebanana.com/file/210204/k25V2B2k.png).

Thanks in advance!
Felix

sfeilmeier, 2021-02-04:
@fabianfnc: Looks strange. Have you seen that before? Any idea?

fabianfnc, 2021-02-10:
Havent seen that before.
Ill try it out myself.

@zoesn89 can you give me your configuration settings?

zoesn89, 2021-02-10:
@fabianfnc sure, but how? screenshot or is there a file, which I can send you? ;)

zoesn89, 2021-02-12:
@fabianfnc here are screenshots:
[Screenshot 1](https://s1.imagebanana.com/file/210212/B5Btc6ln.png)
[Screenshot 2](https://s1.imagebanana.com/file/210212/kLjeWG2w.png)
[Screenshot 3](https://s1.imagebanana.com/file/210212/ieOubJ3v.png)

zoesn89, 2021-02-21:
Hey @sfeilmeier @fabianfnc,
could you find a solution, yet?

fabianfnc, 2021-02-21:
It is hard to tell with that config.
I would try the following things:
- get the latest version of OpenEMS
- go through the getting started and check if the error still occours 
[https://openems.github.io/openems.io/openems/latest/gettingstarted.html](url)
- if the error doesnt occour afterwards, add your old config step by step until the historical data isnt available anymore

zoesn89, 2021-05-17:
> It is hard to tell with that config.
> I would try the following things:
>
> * get the latest version of OpenEMS
> * go through the getting started and check if the error still occours [...]

hey @fabianfnc,

I did what you suggested, but I still have no historical data.

any other ideas?

sfeilmeier, 2021-05-17:
Hi @zoesn89,

there might be an error because of the two timedata providers (Timedata.InfluxDB and Timedata.RRD4j). I am not sure this setup is properly tested. What happens if you disable RRD4j?

After checking possible failures, my assumption is, that RRD4j might not be able to persist data to the hard disk. It writes to the directory configured by `openems.data.dir` in the EdgeApp.bndrun (https://github.com/OpenEMS/openems/blob/develop/io.openems.edge.application/EdgeApp.bndrun#L10). The *Deploy OpenEMS Edge* guide (https://github.com/OpenEMS/openems/blob/develop/doc/modules/ROOT/pages/edge/deploy.adoc#create-a-systemd-service-definition) unfortunately does not show, that this parameter has to be configured on a Linux system. 

See this example service file:

$ cat /etc/systemd/system/openems.service
```
[Unit]
Description=OpenEMS
After=network.target

[Service]
User=root
Group=root
Type=notify
WorkingDirectory=/usr/lib/openems
ExecStart=/usr/lib/jvm/java-8-openjdk-armhf/bin/java \
    -Dorg.osgi.framework.storage=/tmp/org.osgi.framework.storage \
    -Dosgi.clean=true \
    -Dfelix.cm.dir=/etc/openems.d/ \
    -Dorg.apache.felix.eventadmin.Timeout=0 \
    -Dorg.apache.felix.http.host=localhost \
    -Dopenems.data.dir=/var/lib/openems/ \
    -XX:+HeapDumpOnOutOfMemoryError \
    -XX:+ExitOnOutOfMemoryError \
    -jar /usr/lib/openems/openems.jar
SuccessExitStatus=143
Restart=always
RestartSec=10
WatchdogSec=180
TimeoutSec=180

[Install]
WantedBy=multi-user.target
```

zoesn89, 2021-05-18:
Hi @sfeilmeier, 

thanks for your answer. Unfortunately, this isn't the problem as it seems -.-

Just to clarify what I've done to deploy my edge device, I'll write down my steps here. Maybe we could find the problem on that way.

1. I downloaded the latest release 2021.9.0 and opened it with eclipse IDE (running with openJDK 8)
2. In the Bndtools Explorer I opened EdgeApp.bndrun under io.openems.edge.application
3. I hit the Export button and exported the app as openems.jar
4. I openend a SSH connection to my BeagleBone black and created the folders /usr/lib/openems, /etc/openems.d and /var/lib/openems (after your last message)
5. My service file /etc/systemd/system/openems.service looks like this:
`[Unit]
Description=OpenEMS 
After=network.target 

[Service]
User=root 
Group=root
Type=notify 
WorkingDirectory=/usr/lib/openems
LimitCORE=infinity
LimitRTPRIO=2
LimitRTTIME=60000000
CPUSchedulingPolicy=rr
CPUSchedulingPriority=1
ExecStart=/usr/bin/java \
    -Dorg.osgi.framework.storage=/tmp/org.osgi.framework.storage \
    -Dosgi.clean=true \
    -Dfelix.cm.dir=/etc/openems.d/ \
    -Dorg.apache.felix.eventadmin.Timeout=0 \
    -Dorg.apache.felix.http.host=localhost \
    -Dopenems.data.dir=/var/lib/openems/ \
    -XX:+HeapDumpOnOutOfMemoryError \
    -XX:+ExitOnOutOfMemoryError \
    -jar /usr/lib/openems/openems.jar
SuccessExitStatus=143 
Restart=always 
RestartSec=10 
WatchdogSec=60 

[Install]
WantedBy=multi-user.target`

6. I executed `systemctl daemon-reload`
7. I transfered the openems.jar file to /usr/lib/openems
8. I executed `systemctl restart openems --no-block; journalctl -lfu openems`
9. After that I configured my edge device over ip-of-my-device:8080/system/console/configMgr and installed 
- Bridge Modbus/RTU Serial
- Controller Api Modbus/TCP Read-Write
- Controller Api MQTT
- Controller Api REST/JSON Read-Write
- Controller Api Websocket
- Controller Balancing Symmetric
- Controller Debug Log
- Controller Ess Fix Active Power
- GoodWe Charger PV1
- GoodWe Charger PV2
- GoodWe ESS
- GoodWe GridMeter
- Scheduler All Alphabetically
- Timedata InfluxDB
- Timedata RRD4J

10. Then, I build the ui with Visual Studio and `ng build -c edge`
11. I copied the files to /var/www/edge/
12. I created a new config file for nginx which looks like this:
`server {
    listen 80;
    server_name openems;
    
    # OpenEMS Web-Interface
    location / {
        root    /var/www/edge;
        index    index.html index.htm;

        error_page    404    300    /index.html;
    }

    # Proxy for OpenEMS REST-Api
    location /rest/ {
        proxy_pass        http://127.0.0.1:8084/rest/;

        proxy_set_header    Host $host;
        proxy_set_header    X-Real-IP $remote_addr;
        proxy_set_header    X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header    X-Forwarded-Proto https;
    }

    # Proxy for OpenEMS Websocket
    location /websocket {
        proxy_pass    http://127.0.0.1:8085;

        proxy_http_version    1.1;
        proxy_set_header    Upgrade $http_upgrade;
        proxy_set_header    Connection "upgrade";
    }
}`

13. After that, I executed `systemctl restart openems --no-block; journalctl -lfu openems`

so far, so good?!

sfeilmeier, 2021-05-18:
Yes, that looks good. Have you tried disabling one Timedata service? And do you have data in InfluxDB for the energy channels?

zoesn89, 2021-05-18:
> Yes, that looks good. Have you tried disabling one Timedata service? And do you have data in InfluxDB for the energy channels?

yes, I disabled RRD4J at first and I disabled influxdb (and enabled RRD4J again) as well. There was no effect at all...

But now I had a look in the dev console on my chrom browser and found this error:
`Timeout while setting current edge       main-es2015.60acdedc9b566b7ab702.js:1`
[see screenshot](https://www.imagebanana.com/s/big/2091/4hpSOUht.png)

zoesn89, 2021-06-01:
@sfeilmeier do you have an idea what it could be?

sfeilmeier, 2024-03-03:
@zoesn89: We try to move all discussions to the OpenEMS Community (https://community.openems.io/), to have them in one central place. Please feel free to open a thread there.

Sorry for "just closing" your issue now. I am just cleaning up.

closed 2024-03-03 by sfeilmeier