OpenEMS/openems #1401: Too greedy slash replacement in MyJsonServer

bantu, 2021-02-27:
### Bug Report or Feature Request (mark with an `x`)
```
- [X] bug report -> please search issues before submitting
- [ ] feature request
```

### Bug description or desired functionality.

In the `public void newSession(UUID sessionIndex, SessionInformation information)` method of `io.openems.edge.evcs.ocpp.server`'s `MyJsonServer` class, we have
```java
String ocppIdentifier = information.getIdentifier().replace("/", "");
```
The resulting string is used as a key in a map
```java
MyJsonServer.this.parent.ocppSessions.put(ocppIdentifier, sessionIndex);
```
and it seems that this map is used to map between data entered in the UI and connections made from charging stations to the OCPP server.

My charging station sends `/ws://192.168.XXX.YYY:9000/CP_1/SERIAL_1234` as the `information.getIdentifier()`. I hence suspect that the `replace()` is only supposed to chop of the first slash on the left hand side. However, since `replace()` replaces *all* slashes, my charging station is not correctly discovered unless I configure `ws:192.168.XXX.YYY:9000CP_1SERIAL_1234` as its OCPP identifier.

bantu, 2021-02-27:
The code mapping from configuration to active session can be found in `OcppServerImpl` and is
```java
UUID sessionId = this.ocppSessions.get(ocppEvcs.getConfiguredOcppId());
```

bantu, 2021-03-01:
Some possibly useful additional information: I don't know why the identifier sent by the charging station starts with a slash in the first place. I only found some mention of *appending a slash* in the [OCPP 1.6 spec](https://raw.githubusercontent.com/mobilityhouse/ocpp/master/docs/v16/ocpp-j-1.6-specification.pdf) in section "3.1.1. The connection URL".

bantu, 2021-04-26:
@sebastianasen Any comments?

sebastianasen, 2021-04-27:
Hi @bantu,
Thank you for your bug report.
Sorry, I was very busy with other things and had holidays in between, so I forgot to answer.

I was using this to avoid wrong configurations. E.g., the IES KeyWatt is always adding one slash at the beginning. 
But of course, replace is too greedy in here and is causing that error.

I hope to find some time to fix it, otherwise you could push your version if you fixed and tested it already with your charger.

bantu, 2021-04-27:
> I was using this to avoid wrong configurations. E.g., the IES KeyWatt is always adding one slash at the beginning.

Okay, that confirms my assumption that this is not the intended behavior. Thank.

> I hope to find some time to fix it, otherwise you could push your version if you fixed and tested it already with your charger.

I can submit a patch, no problem. Basically, the replace operation needs to be replaced with a left-trim operation.

sebastianasen, 2021-04-27:
Yes, that would be great.

bantu, 2021-06-01:
Update: Current implementation removes all slashes. With the previous proposal, only slashes on the left hand side would be removed. I am no longer convinced this is a good idea since this manipulation of `ocppIdentifier` seems arbitrary.

sebastianasen, 2021-06-01:
Since the IES KeyWatt is still adding one slash at the beginning, the only alternative would be, to configure the ocppIdentifier of the IES KeyWatt Components per default with a slash at the beginning - to match with the original.

hydroid7, 2023-03-30:
Hi, there
@bantu could the solution suggested by @sebastianasen solve your issue?
Best

closed 2023-04-11 by hydroid7