OpenEMS/openems #1480: Modbus Registerbelegung 303 falsch?

chk2902, 2021-05-15:
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
```

Hi Stefan,

Ich programmiere eine Android-App für die Ansteuerung und Regelung von Wallboxen (https://play.google.com/store/apps/details?id=com.chk.go_elocal), und ein Kunde hat mich auf OpenEMS gebracht. Eigentlich ein super System. Die Modbus-verbindung klappte auch auf Anhieb, aber ich verstehe einen Wert nicht.

15.5.2021, 16:10:05INFOio.openems.edge.controller.debuglog.DebugLogImpl[ctrlDebugLog0] _sum[State:Ok Ess SoC:100 %|L:1761 W Grid:-1549 W Production:1741 W Consumption:212 W] charger0[P:1741 W] ess0[SoC:100 %|L:1761 W|Allowed:0;10000 W|On-Grid|Read-Only-Mode] meter0[L:-1549 W]

Und zwar ist der Akku voll, und seine PV produziert 1741W, er exportiert 1549W, Hausverbrauch 212W. Alles gut so weit, aber wieso ist der ESS-Wert 1761W? Diesen bekomme ich dann auch per Modbus und sehe dann (plusminus ein paar Sekunden, daher minimal andere Werte):

MERGED: {
  "OpenEMS.ID": 25089,
  "OpenEMS.State": 0,
  "EM.ExportW": 1546,
  "Battery.SoC": 100,
  "Battery.ProductionW": 1752,
  "PV.ProductionW": 1734
}

was für mich keine Sinn ergibt, der Akku darf höchstens ein paar Watt Ladeerhaltung "ziehen".

Register 303, "_sum/EssActivePower", ist laut Glossar "Beladeleistung (negativ) oder Entladeleistung (positiv) in
[W]".

Allerdings ist die Batterieleistung in der Oberfläche bei fenecon in Ordnung (!)... kann es sein, dass das Register 303 eine falsche Zuordnung hat?

Christian

sfeilmeier, 2021-05-16:
Hallo Christian,

danke für das positive Feedback und coole App! Mit der OpenEMS-Api kannst du in deiner App sofort viele verschiedene Stromspeichersysteme am Markt unterstützen. Wir haben auch eine ähnliche Logik als OpenEMS-Controller implementiert, die mit verschiedenen Ladesäulen funktioniert: https://github.com/OpenEMS/openems/tree/develop/io.openems.edge.controller.evcs

Zum zum eigentlichen Problem. Die "_sum/EssActivePower" beschreibt die Leistung auf der AC-Seite des Batteriewechselrichters des Stromspeichersystems. Es gibt zwei unterschiedliche technische Aufbauten von Stromspeichersystemen: AC- und DC-Systeme. Bei einem AC-System ist die "_sum/EssActivePower" auch die (AC-seitige) Be- oder Entladung der Batterie; bei einem DC-System kann darin auch die Photovoltaik-Produktion enthalten sein. Hier ist das ganz anschaulich erklärt: https://www.memodo.de/blog/pv-speichersysteme-unterschied-ac-und-dc/. Die Beschreibung in OpenEMS ist leider noch nicht optimal, das nehme ich mal mit.

Die tatsächliche Be- oder Entladung der Batterie findest du im Channel `_sum/EssDischargePower` (siehe https://openems.github.io/openems.io/javadoc/io/openems/edge/common/sum/Sum.ChannelId.html#ESS_DISCHARGE_POWER); ich stelle aber gerade fest, dass der anscheinend noch nicht auf ein Modbus-Register gemappet ist (das wäre hier: https://github.com/OpenEMS/openems/blob/e1500a46536077503410173fb87f2be88b6239f7/io.openems.edge.common/src/io/openems/edge/common/sum/Sum.java#L586).

Wenn du mir (gerne per privater E-Mail) die FEMS-Nr. des Kunden schickst, kann ich dort evtl. kurzfristig eine Testversion mit dem aktivierten Register aufspielen.

Gruß,
Stefan

chk2902, 2021-05-16:
Achso, und ab welcher Firmware Version ist das korrigiert ( damit ich eine 
entsprechende Hinweisbox anzeigen kann)?

Lg,

Christian

sfeilmeier, 2021-05-17:
Aktuell gibt es noch kein Update, das dieses Problem behebt. Ich werde den zugehörigen Pull-Request dann hier verlinken.

zoesn89, 2021-05-19:
> Wenn du mir (gerne per privater E-Mail) die FEMS-Nr. des Kunden schickst, kann ich dort evtl. kurzfristig eine Testversion mit dem aktivierten Register aufspielen.

@sfeilmeier das würde mich auch sehr interessieren. Kannst du eventuell kurz zeigen, wie man solche Register mappen kann?

Danke und schöne Grüße
Felix

chk2902, 2021-05-27:
Moin Stefan,

gibt es hier schon was Neues? Mein Kunde fragt nach, denn er möchte jetzt langsam die PV-geregelte Ladung machen, und dazu muss er den Batterieladungsstrom bekommen können über Modbus. Und dazu fehlt ihm dieses Register.

Grüße,

Christian

zoesn89, 2021-06-08:
Hey, gibts hier schon ein Update? Warte auch sehnsüchtig darauf ;)

chk2902, 2021-06-09:
Nope. Mehr als Nachhaken kann ich auch nicht ;( Ruft doch mal den Support an...

venu-sagar, 2021-06-14:
Hello Christian,

I have added the ESS_DISCHARGE_POWER channel and you can find the branch [here](https://github.com/OpenEMS/openems/tree/feature/modbusbugfix). 
Now you can find the actual charging and disachring power of the battery from the register address 415. remember, 'negative' values for Charge and 'positive' for Discharge.

Mapping of channel to the modbus register is pretty straight forward which you can find it here, https://github.com/OpenEMS/openems/commit/56fc088848b2eab4963d7fd5e52c85c68e1e34ee#:~:text=.channel(113%2C%20ChannelId.ESS_DISCHARGE_POWER%2C%20ModbusType.FLOAT32)%20//

sfeilmeier, 2021-06-15:
@venu-sagar Can you please open a Pull-Request for it?
@chk2902 Mir fehlt noch die Info, um welche FEMS-Nr es geht. Dann würde ich dort das nächste Release priorisiert aufspielen lassen.

zoesn89, 2021-06-22:
Jetzt muss ich mal doof fragen: Nun wurden die drei Werte von Modbus RTU auf die Channels gemappt. Aber kann ich die Werte nun auch über Modbus TCP (also über die Bridge) abfragen oder müssen die dort auch noch gemappt werden?

sfeilmeier, 2021-06-23:
@zoesn89: @venu-sagar  hat die `ModbusSlaveNatureTable` erweitert. Diese definiert, welche Register von einem OpenEMS Edge System über den Modbus-Api-Controller gelesen werden können. (https://github.com/OpenEMS/openems/tree/develop/io.openems.edge.controller.api.modbus) - d.h. externes Gerät wie z. B. die Android-App von @chk2902 liest Werte von OpenEMS Edge aus

Das hat mit der Modbus-Bridge nichts zu tun. Die Modbus-Bridge wird verwendet, um Daten von einem externen Gerät in OpenEMS Edge einzulesen - d.h. OpenEMS Edge liest Daten von einem externen Gerät, z. B. einem Zähler.

Habe ich die Frage richtig verstanden und beantwortet?

zoesn89, 2021-06-23:
@sfeilmeier ah, ok. Dann verstehe ich das vielleicht falsch:
Ich dachte, der Modbus RTU service sammelt die Daten per Draht ein und die Bridge stellt die Daten dann per TCP über die API bereit ;)
So habe ich die Daten z.B. über den ModbusTCP Adapter in mein ioBroker System einlesen können. Jetzt fehlt mir quasi dazu noch die passenden Register. Verstehst du was ich meine? ;)

Vielleicht kommen die Werte über die Adresse 301 ESS: Active Power nun auch richtig an, da das Mapping nun passt. Zuvor hatte ich nämlich immer nur positive werte und nur negative bis -40W, wenn die Batterie im Stand-By war.

P.S.:
Wollte gerade das neue Release einspielen, hatte dann aber mächtig Probleme die Module Charger1 und 2 wieder in Gang zu bekommen und musste erstmal wieder downgraden.

closed 2021-10-07 by sfeilmeier