OpenEMS/openems #1481: Problem with MQTT

zoesn89, 2021-05-17:
```
- [x] bug report 
- [ ] feature request
```

Hey everyone, 

thank you for implementing MQTT. I think, this is a game changer for integrating openEMS with other smart home systems.

Unfortunately, I’m not able to get it working. I’m using mosquitto as a mqtt broker and enabled websockets in my config file on port 9001. On my edge device, I tried to get it working with ws://ip-of-my-broker:9001, but I get the following error in the log:

12.5.2021, 09:38:11 INFO io.openems.edge.controller.api.mqtt.SendChannelValuesWorker [ctrlControllerApiMqtt] Error while sending MQTT topics: ctrlControllerApiMqtt/_PropertyClientId, ctrlControllerApiMqtt/State, ctrlControllerApiMqtt/RunFa...

I also tried a tcp connection via Port 1883, but I got the same error.

What could be the problem?

sfeilmeier, 2021-05-17:
Hi @zoesn89,

there should be a more detailed error message somewhere in the logs. Unfortunately I just realized, that it is only a plain `System.out.println`, but maybe you can still find it. 

Example:
```
2021-05-17 20:52:50,244 [Worker-0] INFO  [i.mqtt.SendChannelValuesWorker] [ctrlControllerApiMqtt] Error while sending MQTT topics: _simulator/_PropertyCapacity, _simulator/_PropertyChargeMaxVoltage, _simulator/_PropertyMinCellVo...
Mon May 17 20:52:51 CEST 2021: Verbindung zu Server kann nicht hergestellt werden
```

Here is the code:
https://github.com/OpenEMS/openems/blob/develop/io.openems.edge.controller.api.mqtt/src/io/openems/edge/controller/api/mqtt/MqttConnector.java#L53

zoesn89, 2021-06-01:
Ok, I tested a lot of things now, but I can't get it working. Did you try it with a mosquitto broker yet?

closed 2023-10-16 by hydroid7