OpenEMS/openems #1507: Outdated tutorial

stefansc1, 2021-06-08:
### Bug Report or Feature Request
```
- [x] bug report
- [ ] feature request
```

### Bug descriptions

I followed the [tutorial](https://openems.github.io/openems.io/openems/latest/gettingstarted.htm) and wanted to give feedback.

#### 1. Download the source code
No problems.

#### 2. Setup Eclipse IDE for OpenEMS Edge and Backend
Apart from the outdated Java version, no problems. I am using the existing openjdk 11 instead.

#### 3. Run OpenEMS Edge and start Simulator
Step 2b (configure a scheduler): the cycle time can not be set in the UI.

Step 2c (setup Controller Debug Log): there are more options (Condensed output, Additional Channels and Ignore Components) which are not shown in the example. Defaults apply. Debug Log info shortened to `[ctrlDebugLog0] _sum[State:Ok]` without any power output.

Step 2d (Simulator Datasource): Time-Delta text input and Realtime checkbox merged into one field. Using default of -1 (not realtime).

:exclamation:Step 2e (Simulator Grid Meter Acting): MinPower/MaxPower must be set at creation, otherwise exceptions are thrown: `java.lang.IllegalArgumentException: Unable to set value [] for Channel [meter0/_PropertyMinActivePower]: For input string: ""`
I am using values of 0 for both. Output looks close to example, but without SoC: `_sum[State:Ok Grid:502 W Consumption:502 W] meter0[502 W]`

Step 2f (Simulator EssSymmetric Reacting): Datasource-ID and Datasource target filter can not be set anymore.

Step 2g (Controller Balancing Symmetric): Max power adjustment rate can not be set anymore.

#### 4. Setup Visual Studio Code for OpenEMS UI
Instead of using Visual Studio Code, I use the command line directly. No problems.

#### 5. Run OpenEMS UI / 6. Integrate OpenEMS Backend
Straightforward frontend stuff, no problems.

#### Implementing a device
1.1 (Create a new OSGi Bundle)
Could also be done directly from the menu (without Other...) but okay.
JRE: as shown, setting execution environment JRE to (outdated?) JavaSE-1.8.
After setting project name and JRE, there is no next screen with settings, only Finish.
After Finish, there are several errors in the new component, which go away in later steps.

1.2 (Define Bundle dependencies)
slf4j.api is no longer part of the buildpath.

1.3 (Define configuration parameters), step 2c (add Meter attribute)
Meter must be imported. Later on, it is explained how to resolve import errors, but this is probably the earliest a user might stumble across one.

1.4 (Implement the OpenEMS Component)
Under src/<component name> there is Config.java, MyModbusDevice.java (an interface defining the enum _ChannelId_) and MyModbusDeviceImpl.java (the actual class). For this to work, the class constructor must be changed (as hinted at by Eclipse). It's a good idea to fix this error before renaming. I renamed both files.
In the class constructor, the last channel line then reads `MeterSimulated.ChannelId.values()`, using the enum from the interface.
Step 7 (set debugLog output): `getActivePower()` already returns an integer and therefore has no value() method.
In the notes regarding the file content, points 18 and 19 are inside the block of point 17.

1.5 (Start the device simulator)
In the test subfolder of the new component there are two files, MyConfig.java and MyModbusDeviceTest.java. The first has an error (The type MyConfig must implement the inherited abstract method Config.type()). Resolved by adding unimplemented methods, which adds the type() method (per default just returns null). I have the impression this never gets used during the tutorial.
Starting the ModbusSlaveSimulator is straightforward and poses no problems.

1.6 (Enable Component) 
:exclamation:Resolving fails because of unmet requirements. Removing all `io.openems.edge.evcs` entries from the run requirements fixes this.
You need to save before resolving, otherwise the new component is not part of the runbundle.

:exclamation:1.7 (Run the implementation)
Adding the new simulated meter throws an error: `bundle io.openems.edge.meter.simulated:<version> (159)[Meter.Simulated(15)] : Invalid syntax in target property for dependency Modbus to `
According to [an earlier answer](https://community.openems.io/t/error-loading-module-invalid-syntax-in-target-property-for-dependency-modbus-to/364) this is normal and expected.
However, the simulated meter fails to connect to the slave simulator. While debugging, I can see the ModbusSlaveSimulator is running. In the console, I get the warning `[bridge.modbus.api.ModbusWorker] [modbus0] FC3ReadHoldingRegisters [meter0;unitid=1;ref=1000/0x3e8;length=1] execution failed: Connection to [127.0.0.1] failed: Verbindungsaufbau abgelehnt (Connection refused)` and the debug log shows `_sum[State:Fault] meter0[L:UNDEFINED] modbus0[State:FAULT: LAN Communication to external device failed]` followed by `_sum[State:Ok] meter0[L:UNDEFINED]` after the bridge is invalidated after the error.

### Software versions used
- OS: Ubuntu 20.04
- Java: openjdk 11.0.11
- Eclipse: 2019-12 (4.14.0)
- no Visual Studio Code
- Angular CLI: 12.0.2
- Node: 14.17.0
- Package Manager: npm 6.14.13
- OpenEMS: current master branch, latest release version 2021.10.0

waqarrashid33, 2022-02-22:
I faced similar issue saying `[bridge.modbus.api.ModbusWorker] [modbus0] FC3ReadHoldingRegisters [meter0;unitid=1;ref=1000/0x3e8;length=1] execution failed: Connection to [127.0.0.1] failed: Verbindungsaufbau abgelehnt (Connection refused)`

I resolved it by using another port instead of 502. I used port 2502 and then it started working. My operating system is also Ubuntu 20.04 so it might be a problem with Ubuntu.

stefansc1, 2023-04-12:
I did not need to implement a new device since, so didn't follow the tutorial any further. The commit does not seem to touch on Getting Started, so section 3 (Run OpenEMS Edge and start Simulator) is still outdated. In my mind, this section is more important than implementing a new device.
As I understand it, the commit was done on develop and has a release tag of 2023.4.0? In any case, the online documentation is still unchanged, which makes it hard to say for sure.

JimsterCoder, 2023-07-20:
I am a new user, just going through the Getting Started and it's frustrated to come to a component that does not exist. Controller Balancing Symmetric. 
Now what do I do?

sfeilmeier, 2023-07-21:
@JimsterCoder: Yes, `Controler Ess Balancing` is the correct replacement for `Controller Balancing Symmetric`. 

I understand, that your experience has been frustrating. OpenEMS is an open source community project, so good documentation relys on people contributing to it. It would be great of you could start a Pull-Request and fix the issue in the docs. Thanks for considering! Also there is OpenEMS Community Forum if any other questions arise in future: https://community.openems.io/

JimsterCoder, 2023-07-26:
A more concerning problem with the tutorial I have run into.  
This is the UI I get.
[image]

this is what it is supposed to look like.
[image]

I have spend an hour or so and can't figure out how to add the grid to the pretty picture, or why it isn't there in the first place.

ALSO, the display is in German and I haven't been able to figure out how to change it to English... or any other language.

JimsterCoder, 2023-07-27:
Above it says

"In the test subfolder of the new component there are two files, MyConfig.java and MyModbusDeviceTest.java. The first has an error (The type MyConfig must implement the inherited abstract method Config.type()). Resolved by adding unimplemented methods, which adds the type() method (per default just returns null). I have the impression this never gets used during the tutorial."

This is the code to add to MyConfig.java, below that last }

	@Override
	public MeterType type() {
		// TODO Auto-generated method stub
		return null;
	}

I put this here to help me should later try to fix the tutorial.... for now I need to move in

UmerHayyatVE, 2023-08-22:
Hi, 
I am trying to configure this on mac, despite having follow the steps [mentioned](https://openems.github.io/openems.io/openems/latest/gettingstarted.html#_integrate_openems_backend) I am getting  the following error
need help here as I spent alot of time to figure it out but no success so far
`WebSocketSubject.js:91 WebSocket connection to 'ws://localhost:8085/' failed: `

JimsterCoder, 2023-08-22:
> Hi, I am trying to configure this on mac, despite having follow the steps [mentioned](https://openems.github.io/openems.io/openems/latest/gettingstarted.html#_integrate_openems_backend) I am getting [...]

Have turned the firewall off on the Mac?

hydroid7, 2023-08-22:
Firewalls to localhost should not matter. Can you please verify that the websocket server is running? You can run `netstat -tulpn | grep 8085` to see if a process is bound to the port.

commit 52d38cc1 (2023-08-25): Docs: Update Getting Started & Implementing a Device (#2331)

- Update text and images
- Fix Bnd templates

Fixes #1507
closed 2023-08-25 by sfeilmeier