OpenEMS/openems #1540: Implement the "Go-e charger" as Charger-App

nicoketzer, 2021-06-22:
### Bug Report or Feature Request (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [x] feature request
```

## What would like to see implemented?

Please Implement the "Go-e Charger" as an OpenEMS app. Go-e provides an open API interface (HTTP or MODBUS TCP) --> https://github.com/goecharger/go-eCharger-API-v1
Via this API data can be read out (voltage, current, power, ...) as well as data can be set (e.g. charging power to be used).

## What is the usecase?

It would be easier to pair the GO-e charging station with OpenEMS. Currently (my personal solution) I read data like battery level via the OpenEMS REST API, process it via Raspberry PI and then pass this data via the GO-e API to the charging station (e.g. charging current). Via an OpenEMS APP, this issue could be bypassed and settings could be made directly in OpenEMS (e.g. charging power at 100% storage, charging power at 80% storage, reduction of charging power (or stop charging if necessary) if with Goodwe ET 10k inverter >10kW would be drawn at emergency power, etc.).

hogmoff, 2021-06-22:
Hi, 
i'm still working and using the Go-e API with OpenEMS. Yet, not found enough time to understand and remove a strange behaviour when i pull the plug into the car the state is changing from loading to waiting.

[image]

If anyone want to help me to fix this issue, of course i'll make a pull request with the go-e api.

hogmoff, 2021-06-22:
I'm opened a pull request #1541 to solve the issue together

sfeilmeier, 2021-08-05:
#1541 has been merged and Go-e Charger is now available in OpenEMS.

closed 2021-08-05 by sfeilmeier