OpenEMS/openems #1544: Multiple errors after update on latest release 2021.11.0

zoesn89, 2021-06-24:
### Bug Report or Feature Request (mark with an `x`)
```
- [x ] bug report
- [ ] feature request
```

### Bug description or desired functionality.
I have multiple errors after updating to the latest release 2021.11.0. I'm using a Goodwe WR and BYD Battery.

`24.6.2021, 12:04:00 INFO io.openems.edge.controller.debuglog.DebugLogImpl
[ctrlDebugLog0] _sum[State:Ok Ess SoC:39 %|L:-1785 W Grid:38 W Consumption:-1747 W] charger0[L:UNDEFINED] charger1[L:UNDEFINED] ess0[SoC:39 %|L:-1785 W|On-Grid|Allowed:0;0 W] meter0[L:38 W]`

`24.6.2021, 12:03:59 ERROR io.openems.common.worker.AbstractWorker
Worker error. NoSuchMethodError: java.nio.ByteBuffer.rewind()Ljava/nio/ByteBuffer;`

`24.6.2021, 12:03:59 ERROR io.openems.edge.timedata.influxdb.InfluxTimedataImpl
[influx0] Unable to write to InfluxDB: partial write: field type conflict: input field "influx0/_PropertyIsReadOnly" on measurement "data" is type integer, already exists as type string dropped=1 for data _componentManager/ConfigNotActivated=0i,_componentManager/DefaultConfigurationFailed=0i,_com...`

I downgraded to the previous release and everything is working again. Do you have ideas?

sfeilmeier, 2021-06-28:
The `rewind()` problem is most likely again related to wrong Java settings, see https://community.openems.io/t/probleme-beim-aufsetzen-edge-goodwe-byd/407/2?u=stefan.feilmeier

The InfluxDB issue comes from the fact, that in earlier versions of OpenEMS the type of a property was guessed. Since #1463 this information is extracted from `MetaTypeInformation`. But in your case that measurement had already been added to the InfluxDB schema using a wrong data type 'string' before and it does not allow to ever change this information.

In the InfluxDB Backend bundle these cases are handled dynamically. This feature is missing in the Edge InfluxDB bundle (see https://github.com/OpenEMS/openems/blob/develop/io.openems.backend.timedata.influx/src/io/openems/backend/timedata/influx/FieldTypeConflictHandler.java)

zoesn89, 2021-06-30:
Ok, but I'm building the JAR-File with the same settings as before. Do I need to upgrade the Java version?
Could this be the problem for charger0/1 as well?

So I have to set up a new database to use influxDB again. Sad, but its ok for me.

hydroid7, 2023-03-29:
Hi @zoesn89 ,
were you able to solve the problem by upgrading to newer Java version?
BTW, we just arrived at Java 17 🥳

closed 2024-01-04 by hydroid7