OpenEMS/openems #1591: When UI service worker registration fails, the UI blocks in an undefined state

clehne, 2021-08-13:
### Bug Report or Feature Request (mark with an `x`)
```
- [X ] bug report -> please search issues before submitting
- [ ] feature request
```

### Bug description or desired functionality.

In current version of Firefox (Win 10) in some situations the OpenEMS UI blocks loading. It stays in an undefined state and a user can't leave this state. When this happens the console output is as follows:
```
Service worker registration failed with: DOMException: The operation is insecure.
```
It may have to do with my privacy settings. I have tried some suggestions from the following [bugzilla ticket](https://bugzilla.mozilla.org/show_bug.cgi?id=1429714). But none helped for me. 

Independent of that, the serviceWorker is registered in app.module.ts:52 without any check if it is done correctly.
Wouldn't it be good to check if the registration was successful or if it failed? An example is shown [here](https://developers.google.com/web/fundamentals/primers/service-workers).

Just to note: Every Firefoy user, which enables the following option
```
cookies set to "Keep until I close firefox"
```
may have problems with the serviceWorker registration. 

I would think, that it would be good to delete the cookie and present the login form, when service worker registration failed. But my angular programming skills are not that good.  Any suggestions what I can do to fix this?

sfeilmeier, 2021-08-16:
I don't think we currently use any features that require the ServiceWorker. Does it work if you simply delete the line?

https://github.com/OpenEMS/openems/blob/develop/ui/src/app/app.module.ts#L52

clehne, 2021-08-20:
Thanks for the hint. I removed the service worker line. OpenEMS UI still works fine without service worker, so the line may be removed completely. I close the ticket. Thanks

Unfortunately my problem still exists, but we are unable to reproduce it on any other computer than mine. It looks like it has to do with my firefox profile settings, interfering with the websocket.

closed 2021-08-20 by clehne