OpenEMS/openems #1657: UI History Simulation MacOS

iHamzaaK, 2021-11-04:
### Bug Report or Feature Request (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [x ] feature request
```

### Bug description or desired functionality.

sfeilmeier, 2021-11-04:
Hi @iHamzaaK. Thanks for getting in touch. The documentation for that feature is unfortunately particularly bad, and I have not worked with it for quite a while, sorry.

According to the code, the file should be located in your home directory, received via `System.getProperty("user.home")` in Java. That should be e.g. `/home/iHamzaaK` on your MacOS/Unix system. 

See the source code here:
https://github.com/OpenEMS/openems/blob/develop/io.openems.edge.simulator/src/io/openems/edge/simulator/timedata/SimulatorTimedata.java#L155

For general questions about OpenEMS I recommend the OpenEMS Community board at https://community.openems.io/

Regards,
Stefan

iHamzaaK, 2021-11-05:
Hey @sfeilmeier,

Thank you for the reply. I tried your way and added the timedata.csv file in that directory. However I still see empty graphs.

Do I need to post this in the community board or you can guide me in this thread

Regards,
Hamza

hydroid7, 2023-03-28:
Hi,
it is advised to post in the community forum for errors.
You also can post there a screenshot or the log of your browser console output. 
This way, we get more information and can help you very efficiently 😃 👍 
Best
Lorant