OpenEMS/openems #1802: openEMS UI connecting to Edge: Cannot login (Timeout while setting current edge)

obel1x, 2022-04-18:
### Bug Report or Feature Request (mark with an `x`)
```
- [ x ] bug report -> please search issues before submitting (done)
```

### Bug description or desired functionality.

I have set up openEMS as written in the [startup guide](https://openems.github.io/openems.io/openems/latest/gettingstarted.html#_setup_eclipse_ide_for_openems_edge_and_backend)

Apache Felix/openEMS Edge is working and websocket is opened, as to see in debug logs:
```
2022-04-18T12:17:08,962 [a4c278c)] INFO  [onent.AbstractOpenemsComponent] [ctrlApiWebsocket0] Activate Controller.Api.Websocket
2022-04-18T12:17:08,962 [a4c278c)] INFO  [socket.AbstractWebsocketServer] Starting [Websocket Api] websocket server [port=8085]
```

Also the UI is starting at port 4200 in visual studio code, but if i am logging in, nothing is happening.

in the browsers debugger i can see: 
```
Websocket connecting to URL [ws://localhost:8085]
websocket.ts:76 Websocket connection opened
wsdata.ts:28 Request      [authenticateWithPassword]
service.ts:172 Timeout while setting current edge
(anonymous) @ service.ts:172
timer @ zone.js:3128
push.23484._ZoneDelegate.invokeTask @ zone.js:443
onInvokeTask @ core.mjs:25535
push.23484._ZoneDelegate.invokeTask @ zone.js:442
push.23484.Zone.runTask @ zone.js:214
push.23484.ZoneTask.invokeTask @ zone.js:525
ZoneTask.invoke @ zone.js:514
data.args.<computed> @ zone.js:3108
```
This affects firefox 99.0 and chromium (100.0.4896.88 (openSUSE Build) stable (64-Bit))

I can see successfull logins in eclipse.
```
2022-04-18T12:32:59,088 [et Api-1] INFO  [edge.core.user.UserServiceImpl] Authentication successful with password only for user [Guest].
2022-04-18T12:32:59,088 [et Api-1] INFO  [roller.api.websocket.OnRequest] [ctrlApiWebsocket0] User [guest:Guest] connected.
```

Maybe this is due to the Websocket listening on tcpv6 by Apache, but UI connects to tcpv4:
Before launch of opemEMS UI - the socket listens to ipv6 only:
```
# netstat -npa | grep 8085
tcp6       0      0 :::8085                 :::*                    LISTEN      17631/java
```
After openems UI is launched:
```
tcp        0      0 127.0.0.1:52146         127.0.0.1:8085          ESTABLISHED 7604/firefox        
tcp6       0      0 :::8085                 :::*                    LISTEN      17631/java          
tcp6       0      0 127.0.0.1:8085          127.0.0.1:52146         ESTABLISHED 17631/java
```
But maybe this is normal.

What can be the cause of this behaviour? How to debug/configure the websocket better?

obel1x, 2022-06-25:
For me there were many reasons why it did not work. this it was my Browser refusing to connect to ws without encryption. I needed to setup Apache with my SSL configuration and do reverse proxy to the WS port.
Unfortunatelly the Browser would not tell this, i found it deep in debug- mode.
One other important thing is, that the webui- source maybe needs to be patched for getting he right WS Adresse to work

obel1x, 2022-06-25:
Leaving this open, there should be some Warning, maybe testing Webbrowsers capabilities before running into silent timeout.

hydroid7, 2023-03-28:
Unsecured websocket connections are only allowed to `localhost` or `127.0.0.1`, otherwise you need  a reverse proxy.
Closing this, since problem solved.

obel1x, 2023-03-28:
So why is it hadcoded the other way? This has not been solved nor is there any improvement.