OpenEMS/openems #1834: UI (anglar) fails on Node JS 18 LTS

zoernert, 2022-05-18:
### Bug description 
`package.json` does not limit node-js engines (eq. v12,v14,...). The latest release is v18 and it will not work with specified angular version. Causing several typedef errors like:
```
Error: node_modules/@angular/common/common.d.ts:23:10 - error TS2724: '"rxjs"' has no exported member named 'Subscribable'. Did you mean 'Subscriber'?

23 import { Subscribable } from 'rxjs';
            ~~~~~~~~~~~~

Error: node_modules/@angular/common/common.d.ts:24:10 - error TS2305: Module '"rxjs"' has no exported member 'SubscriptionLike'.

24 import { SubscriptionLike } from 'rxjs';
            ~~~~~~~~~~~~~~~~

``` 

Recommendation: Set engines constrain in package.json until fixed.

sfeilmeier, 2022-05-18:
@venu-sagar: Could you please take care of this?

venu-sagar, 2022-05-19:
There seem to have bugs if we use the engine version above 17.
with v16.15 to be the latest LTS version, I am adding an engine constraint to never allow beyond v16.15 for now. This needs to be changed/updated.

made changes in #1835

commit 4aad6cf6 (2022-05-19): UI: Constraint on node engine version (#1835); Closes #1834

* Added a constraint for node engine version to avoid bugs during compiling.

* Added comments for clear understanding.
closed 2022-05-19 by venu-sagar