OpenEMS/openems #1926: Feature Request; Elspotprices for the Nordics

hansenelectric, 2022-09-13:
### Bug Report or Feature Request (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [x] feature request
```

### Bug description or desired functionality.

I would like to have the Day ahead spotprices in DK and neighboring countries implemented like the Tibber, aWattar, Stromdao functionality. It should however also be possible to enter a variable DistributionSystemOperator costs on top, which in Denmark is different between summer and winter (is regulated every year) and the operating hour, eg. [in DKK/kWh]
Winter tarif (oct-mar);
low load (24-06); 0,2126
high load (6-17 +21-24); 0,6379
Peak load (17-21); 1,9135

Summer tarif (apr-sep)
low load (24-06); 0,2126
high load (6-17 +21-24); 0,3189
Peak load (17-21); 0,8291

-->
https://www.energidataservice.dk/tso-electricity/Elspotprices
https://radiuselnet.dk/elnetkunder/tariffer-og-netabonnement/prisaendringer/

hydroid7, 2023-03-28:
Hi,
thank you for the feature request. Can you please open a thread in the OpenEMS Community to discuss about your idea?
Best

hansenelectric, 2023-03-28:
Hi
For sure. The thread is here; https://community.openems.io/t/feature-request-elspotprices-for-the-nordics-1926/1466

hydroid7, 2023-03-30:
Closing this here because of community.

closed 2023-03-30 by hydroid7