OpenEMS/openems #1933: Show total battery cycles

benzman81, 2022-09-21:
### Bug Report or Feature Request (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [ x ] feature request
```

### Bug description or desired functionality.
Would be awesome to see the total battery cycles in the battery widget. This can give a hint of the battery lifetime.

Sn0w3y, 2023-09-03:
The Battery Lifetime can be read mostly out of the SoH of the Battery itself i guess

sfeilmeier, 2024-03-03:
@benzman81: We try to move all discussions to the OpenEMS Community (https://community.openems.io/), to have them in one central place. Please feel free to open a thread there.

Sorry for "just closing" all of your issues now. I am just cleaning up.

closed 2024-03-03 by sfeilmeier