OpenEMS/openems #1956: Implement Shelly 1PM

hansgeorg1972, 2022-10-25:
### Bug Report or Feature Request (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [x] feature request
```

### Bug description or desired functionality.
It would be cool if the Shelly 1 PM would be implemented as a Meter and Relay.
Also the Shelly Family which is supported by openems would grow (at the moment Shelly-Plug(s) and Shelly 2.5 are supported)

The big advantage of the 1 PM Shelly would be the higher Power-Rate so it can be used to monitor (cheeply) biger loads like kitchen devices(e.g. Shelly Plug (S) can handle up to 2500W and the 1 PM can handle up to 3500W).

The API the 1 PM Shelly (and all other Gen.2 Shelly´s use) can be found here:
https://shelly-api-docs.shelly.cloud/gen2/General/RPCProtocol
or specific:
https://shelly-api-docs.shelly.cloud/gen2/Devices/ShellyPlus1PM

Also i attached a Datasheet for the Shelly 1 PM
[Datenblatt_Shelly_1PM.pdf](https://github.com/OpenEMS/openems/files/9863411/Datenblatt_Shelly_1PM.pdf)

nicoketzer, 2023-03-12:
Hello there,
I just wrote with Shelly. 

I don't know why the 1PM shows up as a Gen.2 device. Shelly support has also assured me that this is a device that can only use the Gen.1 API.

Without further ado, I have tested this with the device. The Shelly 1 PM can be used and accessed normally via the "Shelly Plug" implementation. Thus, the whole thing should have settled.

On my installation of OpenEMS there is now connected a Shelly-1PM via the Shelly-Plug Impl. of OpenEMS.

Sn0w3y, 2023-09-01:
https://github.com/Sn0w3y/shellyGen2

here you go ;)

sfeilmeier, 2023-09-01:
Should be settled. FYI, there is a thread about this topic and some source code in OpenEMS Community as well: https://community.openems.io/t/hardware-implementation/1822/8

closed 2023-09-01 by sfeilmeier

Sn0w3y, 2023-09-01:
If someone is interested just comment on the post on community.openems.io - i will provide the dirty code ;) Maybe someone wants to clean it up and pull request it? :)