OpenEMS/openems #2091: Set maximum storage level in percent

julhei, 2023-02-28:
### Bug Report or Feature Request (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [X] feature request
```

### Bug description or desired functionality.

As owner of a Fenecon storage solution, which is built on top of OpenEMS, I would like to be able to set a maximum storage level in percent to avoid storage levels above a certain threshold in times when energy requirements are less or daily production is high. This should help the storage system to maintain a high maximum storage capacity over a long period of time (decades) and it contributes to maintaining the high value of the investment.

Currently, I am achieving this charging behaviour by manually setting the mode to manual until 23:59 in the App Grid-optimized charging, when the battery level has reached 80%. I would love to have this automated. I would also be open to accomplish this task using one of the write interfaces (Modbus or REST) out of my home automation system (Home Assistant), or I am open to any other recommendations.

hydroid7, 2023-03-30:
Hi, 
Thank you for that short explanation. 
I have the impression, that you look for a kind of  "battery protection mode".
Feel free to open a thread in OpenEMS community for discussing your idea further. 
Best

sfeilmeier, 2024-03-03:
@julhei: We try to move all discussions to the OpenEMS Community (https://community.openems.io/), to have them in one central place. 

If you miss a feature in your FENECON energy storage system, I recommend getting in touch with FENECON Service (https://fenecon.de/kontakt/). Also note, that for LiFePO4 batteries, it is recommended to charge them "full" regularly, to enable cell balancing and SoC calibration.

closed 2024-03-03 by sfeilmeier