OpenEMS/openems #2124: Optimized Changelog.Component.ts

Sn0w3y, 2023-03-28:
### Bug Report or Feature Request (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [X ] feature request
```

### Bug description or desired functionality.

I optimized the changelog.component.ts so it is configurable without a need to compile it again from the /src/appassets/changelog.json:

`import { Component, OnInit } from '@angular/core';
import { ActivatedRoute } from '@angular/router';
import { TranslateService } from '@ngx-translate/core';
import { environment } from 'src/environments';
import { Service } from '../../../shared/shared';
import { Role } from '../../../shared/type/role';
import { Changelog, Library, OpenemsComponent, Product } from './changelog.constants';
import { HttpClient } from '@angular/common/http';

@Component({
    selector: 'changelog',
    templateUrl: './changelog.component.html'
})
export class ChangelogComponent implements OnInit {

    public environment = environment;

    protected slice: number = 10;
    protected showAll: boolean = false;
    public changelogs: {
        version: string,
        changes: Array<string | { roleIsAtLeast: Role, change: string }>
    }[] = [];

    constructor(
        public translate: TranslateService,
        public service: Service,
        private route: ActivatedRoute,
        private http: HttpClient,
    ) { }

    ngOnInit() {
        this.service.setCurrentComponent({ languageKey: 'Menu.changelog' }, this.route);
        this.loadChangelog();
    }

    public readonly roleIsAtLeast = Role.isAtLeast;
    public numberToRole(role: number): string {
        return Role[role].toLowerCase();
    }

    private loadChangelog(): void {
        this.http.get('assets/changelog.json').subscribe(
            (data: any[]) => {
                this.changelogs = data.map(item => ({
                    version: item.version,
                    changes: item.changes.map(change => {
                        if (typeof change === 'string') {
                            return change;
                        } else {
                            return { roleIsAtLeast: Role[change.roleIsAtLeast], change: change.change };
                        }
                    }),
                }));
            },
            (error) => {
                console.error('Failed to load changelog.json:', error);
            }
        );
    }
}
`

Example changelog.json:

`[
    {
        "version": "x.y.z",
        "changes": [
            {
                "type": "link",
                "text": "OpenEMS Releases",
                "url": "https://github.com/OpenEMS/openems/releases"
            }
        ]
    },
    {
        "version": "1.2.0",
        "changes": [
            "Added support for new battery storage system",
            "Improved user interface for energy consumption monitoring",
            {
                "roleIsAtLeast": "ADMIN",
                "change": "Bug fixes and security updates"
            }
        ]
    },
    {
        "version": "1.1.1",
        "changes": [
            "Fixed minor bugs in energy production calculation",
            "Updated translations for several languages",
            {
                "roleIsAtLeast": "INSTALLER",
                "change": "Performance improvements for API calls"
            }
        ]
    },
    {
        "version": "1.1.0",
        "changes": [
            "Implemented customizable dashboard",
            "Enhanced data visualization with new charts and graphs",
            {
                "roleIsAtLeast": "ADMIN",
                "change": "Updated dependencies to latest versions"
            }
        ]
    },
    {
        "version": "1.0.0",
        "changes": [
            "Initial release of the OpenEMS platform",
            "Basic functionality for energy management and monitoring"
        ]
    }
]`

hydroid7, 2023-03-29:
Hi @Sn0w3y ,
thank you for improving this bit of OpenEMS! 👍 
Could you please open a pull request to give us an overview about your proposed changes?
Thanks!

Sn0w3y, 2023-03-29:
Hi :)

If i am honest i do not know how to do this :o I am a basic Coder and i have more improvements to propose but i dont know how :)

Could you please tell me ?

Thanks !

hydroid7, 2023-03-30:
Sure, we welcome contributions in form of pull requests. 
For that please fork the repo, add your changes and create a pull request.
Best

Sn0w3y, 2023-03-30:
done !

closed 2024-02-14 by Sn0w3y