OpenEMS/openems #2161: Aggregate component channel values available in the same cycle

JoopAue, 2023-05-01:
### Bug Report or Feature Request (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [x] feature request
```

### Bug description or desired functionality.

Hi team,

Virtual component clusters such as `PV-Inverter Cluster` and `Meter Virtual Symmetric Add` are a great addition to OpenEMS.
However, the way they are implemented their computed aggregate values are always running one cycle behind. Channel values read from modbus in for example two Janitza meters are read in Cycle `n`, but the virtual aggregate meter only exposes the aggregate value on its channels on Cycle `n+1`. 

My proposal is an adjustment to the CycleWorker such that the process image of aggregate components is constructed after those of all other components, such that the aggregate channel values are available in the same cycle.

Is this a use-case beneficial to OpenEMS? If so, I'm happy to make and share a proposal, and implement this on agreement. 

Happy to provide extra info!

sfeilmeier, 2023-05-01:
Hi and thanks for opening the discussion.

The way we typically solve this problem nowadays in more modern code is, that we register listeners on the source channels. This updates the target channels 'next value' whenever the source changes, which is then in time for the BEFORE_PROCESS_IMAGE event.

See this snippet as an example:
https://github.com/OpenEMS/openems/blob/develop/io.openems.edge.meter.api/src/io/openems/edge/meter/api/AsymmetricMeter.java#L1028

Does this solve your question?
If you can refactor old code in such a way, it is highly welcomed.

Regards,
Stefan

sfeilmeier, 2023-05-04:
Hi Joop,

Yes, I agree on the downsides of this approach, but from my experience those are negligible, because the calculations are very simple and quick.

Regards,
Stefan

closed 2023-05-04 by JoopAue