OpenEMS/openems #2258: Live-Demo on Gitpod

oke24, 2023-07-07:
- [x ] bug report -> please search issues before submitting

### Bug description or desired functionality.
The Live-Demo on Gitpod (https://gitpod.io/#https://github.com/OpenEMS/openems/tree/develop) is broken. 

The build for openems.edge runs into an error:
```
> Task :io.openems.edge.common:compileJava
Picked up JAVA_TOOL_OPTIONS:  -Xmx3489m
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':io.openems.backend.common:jar'.
> io.openems.backend.common has errors, one error was reported

BUILD FAILED in 1m 3s
321 actionable tasks: 170 executed, 151 up-to-date
```
This results in an error message in the UI:
[image]

Regards.

vangola-silver, 2023-10-02:
The Live Demo is still broken. 
After building the demo in Gitpod I cant login into the Web-App because I get a blank Login-Page:
[image]

In Gitpod-Task 1 I can see following Error message:
`2023-10-02T09:00:42,345 [socket-8] ERROR [socket.AbstractWebsocketServer] [Edge.Websocket] OnInternalError for EdgeWebsocket.WsData [edgeId=edge0]. class java.lang.NullPointerException: Cannot invoke "io.openems.backend.uiwebsocket.impl.WebsocketServer.getConnections()" because "this.server" is null
java.lang.NullPointerException: Cannot invoke "io.openems.backend.uiwebsocket.impl.WebsocketServer.getConnections()" because "this.server" is null
        at io.openems.backend.uiwebsocket.impl.UiWebsocketImpl.sendSubscribedChannels(UiWebsocketImpl.java:234)
        at io.openems.backend.edgewebsocket.OnNotification.handleDataNotification(OnNotification.java:134)
        at io.openems.backend.edgewebsocket.OnNotification.run(OnNotification.java:59)
        at io.openems.common.websocket.OnNotificationHandler.run(OnNotificationHandler.java:22)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
        at java.base/java.lang.Thread.run(Thread.java:833)
`

sfeilmeier, 2023-11-01:
Gitpod demo should be working again properly after todays release.

Thanks for reporting and analyzing the issue!

closed 2023-11-01 by sfeilmeier