OpenEMS/openems #2276: Supports can protocol

luckyfisher, 2023-07-20:
In some business, the function of CAN will be used. Can we add support for CAN?

clehne, 2023-07-20:
Hi luckyfisher, we already have a CAN Bridge which is working almost like the Modbus Bridge. It uses the CAN Socket API. A library for Kunbus RevolutionPi (with CAN extension) is already included. The CAN bridge is not yet published, because we can't provide an Open Source driver, which uses the bridge. This makes it a bit complicated to use the CAN brdige. Nevertheless I will make a PR of the bundle by next week.

sfeilmeier, 2023-07-21:
I know that also @hydroid7 has been working on a CAN implementation. Maybe he can share it as well.

@luckyfisher: the OpenEMS Community Forum might be a better fit for questions like this: https://community.openems.io Could you possibly ask the question again over there? I am closing the issue here.

closed 2023-07-21 by sfeilmeier