OpenEMS/openems #2277: awattar Austria

cf99, 2023-07-20:
### Bug Report or Feature Request (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [x ] feature request
```

### Bug description or desired functionality.

I saw some differences between the German and Austrian prices. Since its one company they use the (hopefully) use the same API. 
Is it possible to change the URL or to select the Country as normal user  in the APP?
Germany --> "private static final String AWATTAR_API_URL = "https://api.awattar.de/v1/marketdata";" 
Austria--> "private static final String AWATTAR_API_URL = "https://api.awattar.at/v1/marketdata";"

sfeilmeier, 2023-07-21:
@venu-sagar Could you please have a look at this?

venu-sagar, 2023-07-24:
@cf99 Thank you for your attention to this matter. We were indeed aware of the market differences, and during the initial implementation, there was no requirement to include the Austrian market API. As a result, the application currently only has a German URL.

cf99, 2023-07-24:
Thanks for the update, so whats next? I really want to change the source. But, since i bought my system and the start up was done by someone else, i cant do a re-Initianization.

KR
Christoph

venu-sagar, 2023-07-24:
@cf99 
To solve your problem, that feature needs to be developed because it is not yet existing. Do you want to develop it? I would be glad to help if you can provide PR with the implementation. Because the development is not in our immediate tasks.
Did you buy the system from somewhere including the EMS? I didn't understand the start-up and re-initialization part!

cf99, 2023-07-24:
@venu-sagar 
Hi, i would really like to contribute something, unfortunately i am no programmer :(
Yes i bought a whole energy-storage system from FENECON. I asked them to simply change form ".de" to ".at", but the response was not so satisfying.
The first "start up/ initialization/  Commissioning/ Inbetriebnahme" was done by an Electrician, so i am just a guest to my own system and not allowed to change anything (important).

venu-sagar, 2023-07-24:
@cf99 Ah okay, I understand completely now. Then I can internally forward this issue to the respective department, this is the preferred of handling any issue. But I cannot guarantee any timeframe result.

cf99, 2023-07-24:
Okay, thanks a lot. 

KR
Christoph

venu-sagar, 2023-07-25:
The issue has been forwarded to the concerned department.

closed 2023-07-25 by venu-sagar