OpenEMS/openems #2302: Optimization AbstractChannelListenerManager  - Clean unused listeners

JoopAue, 2023-08-04:
### Bug Report or Feature Request (mark with an `x`)
```
- [x] bug report (more an optimization)
- [ ] feature request
```

### onSetNext and onChange Listeners are not cleaned up
Hi team,

After diving into OSGI's Declarative Services and understanding the need for the `AbstractChannelListenerManager` I was debugging to see how it works and noticed that the listeners are not removed from internal list that is kept. At some point there were ~500 listeners in the list, and probably 350 unused.

Can we safely remove them to be garbage collected?
See my suggestion in the snippet below.

``` JAVA
public synchronized void deactivate() {
    for (OnSetNextValueListener<?> listener : this.onSetNextValueListeners) {
        Channel<?> channel = listener.component.channel(listener.channelId);
        channel.removeOnSetNextValueCallback(listener.callback);
    }
    this.onSetNextValueListeners.clear(); // NEW
    for (OnChangeListener<?> listener : this.onChangeListeners) {
         Channel<?> channel = listener.component.channel(listener.channelId);
         channel.removeOnChangeCallback(listener.callback);
    }
    this.onChangeListeners.clear(); // NEW
}
```
Also see: https://github.com/OpenEMS/openems/blob/develop/io.openems.edge.common/src/io/openems/edge/common/channel/AbstractChannelListenerManager.java#L52

Happy to submit a PR, but wanted to check first.

sebastianasen, 2023-08-04:
@sfeilmeier Could you have a short look into it

sfeilmeier, 2023-08-04:
Hi, and thanks for bringing this up.

The idea why I did not clear the lists was that I assumed, that the `deactivate()` method will anyways be called only directly before the AbstractChannelListenerManager itself gets destroyed, so it should not be necessary to clear all local variables.

But in fact after your post I found that e.g. in `EssClusterImpl` and in Virtual-Add-Meter (`AddChannelManager`) it is used differently - i.e. calling `deactivate(); activate()` to update the listeners. In that case it makes indeed sense to clear the lists manually.

Thanks & looking forward to reviewing the PR.

closed 2023-08-10 by JoopAue