OpenEMS/openems #2321: Open ems post endpoints not working with /jsonrpc

SaikeshErramshetty, 2023-08-16:
### Bug Report or Feature Request (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [ ] feature request
```

### Bug description or desired functionality.

While trying to exeute anypost request as per the documentation where it is required to hit the post end point via 
http://x:user@localhost:8084/rest/channel/io0/Relay1/jsonrpc it shows

`{"jsonrpc":"2.0","id":"00000000-0000-0000-0000-000000000000","error":{"code":1,"message":"Missing arguments to handle Channel","data":["Missing arguments to handle Channel"]}}`

As to add more to the description this is how my curl request is looking

 ```
curl -X POST -H "Content-Type: application/json" \
    -d '{
  "method":"componentJsonApi",
  "params":{
    "componentId":"ctrlApiModbusTcp0",
    "payload":{
      "method":"getModbusProtocol",
      "params":{

      }
    }
  }
}' \
    http://admin:admin@127.0.0.1:8084/rest/channel/io0/Relay1/jsonrpc
```

Any leads would be highly appreciated.

f2010126, 2023-10-04:
Hi, 
You have used the url  the `rest` endpoint. The query translates into look for the channel Relay1/jsonrpc in device with id `io0`.

Try sending to the url ` http://admin:admin@127.0.0.1:8084/jsonrpc`.

Also, it's a better practice (safer than sending the details in plain text) to include the username and password in the headers. 
You can encode the username and password with base64 encoding (for example). Then add it to the headers as:
```
'Authorization': 'Basic 64_bit_encoding_of_user_pass'
```

sfeilmeier, 2023-10-04:
Hi,

the FEMS documentation (based on OpenEMS) also shows how to use curl, python or the Talend API tester to access the API: https://docs.fenecon.de/de/_/latest/fems/fems-app/OEM_App_REST_JSON.html (sorry, German only right now)

Thanks @f2010126: full ACK with your explanation! 👍

Regards,
Stefan

hydroid7, 2023-10-16:
I recommend this issue be closed since the author received insights about his topic and further discussion can be done in the [community](https://community.openems.io/).