OpenEMS/openems #2360: Historical data not correctly displayed when changing device time zone

flsc42, 2023-09-15:
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
```

I recently opened the history view from a device that was set to Portugal time zone (-1 hour compared to my EMS time zone). 
The historical data of _yesterday_ showed correctly in the graph, but not in the numbers below, instead it showed all "0" (see screenshot).

[image]

Interesting aspects: 
- All graphs and data are shifted to the new time zone. An EMS event happening at 6pm is displayed as happening at 5pm on the device that is on Portugal time.
- Today and all previous days before yesterday were displayed correctly (but shifted). 
- If the device zime zone is set East of the EMS time zone, data is displayed (also shifted).

Expected behaviour: 
- all data and events should be displayed in EMS time, no matter where on the world you view it.

lukasrgr, 2023-10-16:
hi,

do you save the data in the database in UTC time?

sfeilmeier, 2023-10-17:
@flsc42 Thanks for reporting. It's maybe not an issue, but a misunderstanding in the expected behaviour. Because of this I would recommend closing the issue here and continuing the discussion on OpenEMS Community -> https://community.openems.io/

> Expected behaviour:
> all data and events should be displayed in EMS time, no matter where on the world you view it.

In fact the system is designed to store all data in timezone UTC (that's what @lukasrgr was referring to) and visualize it according to the timezone of the viewer - not the EMS timezone.

sfeilmeier, 2023-10-17:
Ok, agree, but this has most likely other reasons. It might be caused by the storage of aggregated timeseries values.

I just realize, that this looks like a FEMS and not OpenEMS screenshot. Is this a issue with FEMS? Then please send a ticket to FENECON Service team -> https://fenecon.de/kontakt/. Thank you.

closed 2023-10-17 by sfeilmeier