OpenEMS/openems #2447: .vscode/settings is forcing format when saving, but editor.defaultFormatter is not specified.

miettal, 2023-11-30:
if a vscode user use another formatter (like prettier) as editor.defaultFormatter in user-side-config, the development will be difficult. because workspace-side-config force to run format and user-side-config formatter try to apply different styling policy.

I think, in case of  forcing format when saving, specifiing workspace policy formatter as editor.defaultFormatter is better.

https://github.com/OpenEMS/openems/blob/edfe1f0ac47a70f25c7c6456289d58344b6c3ab7/ui/.vscode/settings.json#L3

sfeilmeier, 2023-11-30:
@lukasrgr: Should we add `editor.defaultFormatter`?

miettal, 2023-11-30:
now `.eslintrc` is in project directory(/ui) and developper following eslint rule, therefore I think "editor.defaultFormatter": "dbaeumer.vscode-eslint" is suitable.

lukasrgr, 2023-11-30:
@venu-sagar i think i saw this in your config

miettal, 2023-12-14:
I tried to configure it in my local dev environment. it's working. how about this?
https://github.com/girasolenergy/openems-fork/commit/6931885b740c061f2af0218239afbadb1d069d5d

// if this is ok, I can make a PR.

[image]

PR #2470 (2024-01-01): fix: specify eslint as vscode workspace default formatter
close #2447
commit 8687a3e4 (2024-01-01): UI: specify eslint as vscode workspace default formatter (#2470)

Closes #2447
closed 2024-01-01 by sfeilmeier