OpenEMS/openems #2458: gradle re-build fails

miettal, 2023-12-14:
### Bug Report or Feature Request (mark with an `x`)

- [x] bug report -> please search issues before submitting

### Bug description or desired functionality.

It seems re-build(after gradle clean) is failed.

reproducing method:

```
./gradlew build
./gradlew clean
./gradlew build
```

actual behavior:

last build command fails by sybol error when compiling io.openems.backend.common.

```
> Task :io.openems.backend.common:compileJava
/home/runner/work/openems-fork/openems-fork/io.openems.backend.common/src/io/openems/backend/common/jsonrpc/request/GetUserInformationRequest.java:5: error: package io.openems.common.jsonrpc.base does not exist
import io.openems.common.jsonrpc.base.JsonrpcRequest;
                                     ^
/home/runner/work/openems-fork/openems-fork/io.openems.backend.common/src/io/openems/backend/common/jsonrpc/request/GetUserInformationRequest.java:19: error: cannot find symbol
public class GetUserInformationRequest extends JsonrpcRequest {
                                               ^
  symbol: class JsonrpcRequest
```

for reference, I made reproducing CI in my fork-repo.
https://github.com/girasolenergy/openems-fork/actions/runs/7206191185/job/19630540280

what is expected?:

build is successful

sfeilmeier, 2023-12-19:
I do not know enough about gradle to fix this.

@da-Kai: Any idea?

miettal, 2023-12-20:
I'm continuing testing.
Currently I know ...
 - gradle build -> gradle clean -> gradle build is ng
 - gradle build -> gradle clean -> gradle build --refresh-dependencies is ok

this difference using gradle cache or not, so this problem related to gradle cache I think.

da-Kai, 2023-12-20:
Hi @miettal,
i was not able to reproduce this behavior.
 
My local machine:
```
Distributor ID:	Linuxmint
Description:	Linux Mint 21.2
Release:	21.2
Codename:	victoria
```

have you tried to build with `./gradlew build --no-cache --no-daemon` ?

miettal, 2023-12-20:
@da-Kai Thank you trying to reproducing

I used --no-build-cache instead of --no-cache, because my gradlew said `Unknown command-line option '--no-cache'.`.
But the build was failed with same error message of first post.

my environment is ubuntu 22.04.
```
uname -a
Linux taisyo-vaio 6.2.0-37-generic #38~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Nov  2 18:01:13 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

I tested this bug on some versions of openems. and I found this bug started from this commit.
This commit has changing of `gradle.properties` file to disable parallel build.
https://github.com/OpenEMS/openems/pull/2338/files#diff-b1e354954ad76c6075e317dca67f898d68f2b3802ac66f4caa461b7903a58a91

I think Originally there was some dependencies-description-related bug, but it had hided by parallel build. // this is just my expectation.

miettal, 2023-12-21:
It seems it works on gradle 8.5... very strange...

miettal, 2023-12-24:
According **Many Tests & Engineering Result** in @girasolenergy (Thank you @Kouda-Titanium -san!!),
We found this fails caused in gradle caching bug that related to parallel-processing(daemon) and file-locking.
Reproducing of this bug is difficult because this is parallel-processing so its strongly depends on environment.
But we share the condition and reproducing method and mecanizm roughly.

 1. gradlew build => make build-cache that include corrupted file ./.gradle/8.4/executionHistory/executionHistory.{lock or bin}
 2. gradlew clean => remove build output
 3. gradlew build
    - => try to build with using build cache because there are no build output.
    - => :io.openems.common:compileJava make io.openems.common/bin/io/openems/common/*.class from io.openems.common/src/
    - => :io.openems.common:processResources but due to corrupted file executionHistory.{lock or bin}, delete io.openems.common/bin/io/openems/common/*.class
    - => :io.openems.common:processResources make io.openems.common/generated/io.openems.common.jar with no class.
    - => :io.openems.backend.common:compileJava fail by reference error because io.openems.common.jar has no class.

Tnterestingly we found same behavior bug in gradle project issue. this problem fixed in gradle 8.5.
https://github.com/gradle/gradle/issues/20830
And We are doubting this bus is (#2458) same to gradle bug (gradle/issues/20830),
because
 - both bug is related to parallel-processing(daemon) and file-locking.
 - this bug is not reproduced in gradle 8.5

From these result, I think better to upgrade gradle 8.5 firstly.
// Technically We can check processing code of gradle to find root cause, but dont have enough time.

Please correct me if there are any mistakes or additional info. @Kouda-Titanium.

Kouda-Titanium, 2023-12-25:
It can be built correctly by removing `executionHistory.lock` before compiling.
But it may be due to that bug with a daemon and lock files rather than corrupted files.

PR #2471 (2023-12-29): fix: upgrade gradle version 8.4 to 8.5
close #2458

gradle versions up to 8.4 has caching-problem and
In openms, build-clean-build steps fails by last build step.
https://github.com/girasolenergy/openems-fork/actions/runs/7206191185/job/19630540280

This changes fix the problem by updating gradle version 8.4 to 8.5

Test is here. it work.
https://github.com/girasolenergy/openems-fork/actions/runs/7282794004/job/19845714357

See also
 - https://github.com/OpenEMS/openems/issues/2458
 - https://github.com/gradle/gradle/issues/20830
commit 08af7119 (2023-12-29): Upgrade Gradle version 8.4 to 8.5 (#2471)

Closes #2458.

Gradle versions up to 8.4 has caching-problem and in OpenEMS build-clean-build steps fails by last build.
https://github.com/girasolenergy/openems-fork/actions/runs/7206191185/job/19630540280

This changes fix the problem by updating gradle version 8.4 to 8.5
closed 2023-12-29 by sfeilmeier