OpenEMS/openems #2469: aWattar Country AUT - GER

cf99, 2023-12-21:
- [x ] feature request

Unfortunately the missmatch of cost between austria and germany is still not considerd.
I just had a houerly price deviation of 300% (okay that sounds huge, 1.4c€ to ~4.2c€) but still it is annoying. because there is just a change in the domain (.de or .at) required...

https://www.awattar.**de**/services/charts/hourly
https://www.awattar.**at**/services/charts/hourly

venu-sagar, 2024-01-10:
@cf99 Thank you for raising this issue. 

We primarily focused on the German market. I will work on adding a feature that lets you choose between the German and Austrian markets for pricing. This way, you can easily view the right prices for your needs. I'll be submitting the pull request (PR) shortly to make this option available.

In the meantime, you can check prices easily using our ENTSO-E feature. This tool is like a general API for getting day-ahead prices in European power grids. It taps into the ENTSO-E transparency platform, making it a simple and effective way to access pricing information. Feel free to use it while we're working on updates for the Awattar's Austrian market integration.

[io.openems.edge.timeofusetariff.entsoe](https://github.com/OpenEMS/openems/tree/develop/io.openems.edge.timeofusetariff.entsoe)

It retrives prices from ENTSO-E platform. You have to create an account at https://transparency.entsoe.eu/ to get an security token and In the configuration just select the option Austria in the "Bidding Zone".

Let me know if you find any issues while initializing ENTSO-E provider.

Sn0w3y, 2024-01-14:
[awattar.zip](https://github.com/OpenEMS/openems/files/13932284/awattar.zip)

fixed it in here - maybe Review and Push it :)

cf99, 2024-01-16:
i am sorry i can review it (I have now idea how to code).
Who will/ can do this?

Sn0w3y, 2024-01-21:
@venu-sagar

PR #2509 (2024-02-01): FEMS Backports
- [KACO Battery Inverter: quicker control behaviour](https://github.com/OpenEMS/openems/commit/851c94d43193076337895d00e74da06902d44f59)
- [Max-Ever-Value: update config only once every 24 hours](https://github.com/OpenEMS/openems/commit/7fa22cc24b3f65c4c5d91cfa590b7faa0114dbb0)
- [EmergencyController & TimeslotPeakshaving: Set PersistencePriority to…](https://github.com/OpenEMS/openems/commit/2cc72f417bae5f2cf6a47880e277edd02c49d283)
- [ToU: calculate essMaxChargePower from usable Capacity](https://github.com/OpenEMS/openems/commit/35f10c1a1a58a0d377a8b79e5e264e2ec5d95b59)
- [T [... cut]
commit d2ebc677 (2024-02-01): FEMS Backports (#2509)

- [KACO Battery Inverter: quicker control behaviour](https://github.com/OpenEMS/openems/commit/851c94d43193076337895d00e74da06902d44f59)
- [Max-Ever-Value: update config only once every 24 hours](https://github.com/OpenEMS/openems/commit/7fa22cc24b3f65c4c5d91cfa590b7faa0114dbb0)
- [EmergencyController & TimeslotPeakshaving: Set PersistencePriority to…](https://github.com/OpenEMS/openems/commit/2cc72f417bae5f2cf6a47880e277edd02c49d283)
- [ToU: calculate essMaxChargePower from usable Capacity](https://github.com/OpenEMS/openems/commit/35f10c1a1a58a0d377a8b79e5e264e2ec5d95 [... cut]
closed 2024-02-01 by sfeilmeier