OpenEMS/openems #2510: MQTT Client does not reconnect

martingruening, 2024-02-02:
### Bug Report or Feature Request (mark with an `x`)
```
- [X] bug report -> please search issues before submitting
- [ ] feature request
```

### Bug description or desired functionality.
MQTT Client does not reconnect after network connection

Setup: OpenEMS Edge 2024.1 in Docker container
Sending topics to an Mosquitto MQTT broker
Works for weeks without a hickup, but when a network interuption happens (e.g. rebooting the router after an update) the MQTT client does not reconnect to the broker

Docker Log:
```
2024-02-02T13:17:10,476 [Worker-0] INFO  [api.mqtt.ControllerApiMqttImpl] [ctrlControllerApiMqtt] Client is not connected
2024-02-02T13:17:10,476 [Worker-0] INFO  [i.mqtt.SendChannelValuesWorker] [ctrlControllerApiMqtt] Error while sending MQTT topics: scheduler0/_PropertyId [...]
```

The 'Client is not connected' message repeats a lot, but no connection is rebuilt.
Only fix is to restart the Docker container.

Expected behaviour would be to retry to rebuild the connection continously.

Sn0w3y, 2024-02-04:
Only fix is to restart the Docker container.

Have you already tried to deactivate and activate the Controller Api MQTT ? Like this should work aswell :)

Sn0w3y, 2024-02-06:
Hallo @martingruening,

kannst du mir deine Config für dein MQTT mal geben, damit ich meinen fix eventuell testen kann?

Wäre super, da ich mich zwar mit dem Java Code auskenne, aber leider nicht mit MQTT 👯‍♂️

Sn0w3y, 2024-02-06:
@martingruening kannst dir mal meinen Approach ansehen und testen bitte - [PullRequest](https://github.com/OpenEMS/openems/pull/2525)

martingruening, 2024-02-07:
@Sn0w3y Natürlich kann ich testen. Wie komme ich an ein jar-File, das den neuen Code enthält?

Sn0w3y, 2024-02-07:
> @Sn0w3y Natürlich kann ich testen. Wie komme ich an ein jar-File, das den neuen Code enthält?

kannst dir ja mit meinem Code in dem PullRequest selber bauen :)

Hab nur die eine Datei geändert :) https://github.com/OpenEMS/openems/pull/2525

martingruening, 2024-02-12:
Hallo @Sn0w3y - es wäre toll, wenn du mir ein JAR senden könntest, das ich testen kann. Sorry für die späte Antwort.

Sn0w3y, 2024-02-12:
https://www.file-upload.net/download-15268371/openems.jar.html

Versuch die mal und bitte berichte ! :) Sieh dir auch die Logs an, hab mal ein paar mehr Details hinzugefügt um zu Debuggen

Sn0w3y, 2024-02-12:
@martingruening hast du bereits testen können ? :)

Sn0w3y, 2024-02-23:
@martingruening gibt es News ?

martingruening, 2024-02-28:
Entschuldigung. Die Emails von GitHub zu diesem Case sind leider im Spamordner gelandet. Der Change funktioniert - es wird jetzt ein Reconnect zum MQTT Broker durchgeführt. Danke!

sfeilmeier, 2024-03-03:
Discussion is continued in the respective PRs (e.g. https://github.com/OpenEMS/openems/pull/2525) and/or on OpenEMS Community https://community.openems.io/

closed 2024-03-03 by sfeilmeier