OpenEMS/openems #2543: Parse Error io.openems.edge.evcs.keba.kecontact.ReadHandler

dtometzki, 2024-02-15:
### Bug Report or Feature Request (mark with an `x`)
```
- [ X] bug report -> please search issues before submitting
- [ ] feature request
```

### Bug description or desired functionality.
I got a Keba P30 Walbox. Now i see in the Log Parse Errors and the status is: Error
```
15/02/2024, 17:43:39
ERROR
io.openems.edge.evcs.keba.kecontact.ReadHandler
Error while parsing KEBA message: JSON failed to parse [com.google.gson.stream.MalformedJsonException: Use JsonReader.setLenient(true) to accept malformed JSON at line 1 column 9 path $]: report 3
15/02/2024, 17:43:39
ERROR
io.openems.edge.evcs.keba.kecontact.ReadHandler
Error while parsing KEBA message: JSON failed to parse [com.google.gson.stream.MalformedJsonException: Use JsonReader.setLenient(true) to accept malformed JSON at line 1 column 9 path $]: report 2
15/02/2024, 17:43:38
```
What would like to see implemented? nothing
What is the usecase?nothing
-->

Sn0w3y, 2024-02-15:
This issue sometimes happens - i actually do not know why.
Restarting the KEBA Application fixes this most of the time. 

If you are using a FEMS please report this to Fenecon ! :)
If not - maybe you can add some debug statements to your Code and see where the JSON is malformed so we can fix this?

dtometzki, 2024-02-15:
Hello @Sn0w3y,

Many thanks. do you know as owner how i can restart the app in FEMS ?

Best regards
Damian

Sn0w3y, 2024-02-15:
> Many thanks. do you know as owner how i can restart the app in FEMS ?

As i told you - please ask Fenecon Support for this Topic :) They are very helpful in this matter !

You as a owner can only restart the FEMS:

https://fenecon.de/faq/#:~:text=2.,in%20das%20Online%2DMonitoring%20%C3%BCbertr%C3%A4gt.

Sn0w3y, 2024-02-15:
@da-Kai can be closed i guess as it is not directly openEMS related ?
Maybe we should really use JsonReader.setLenient(true) for Kebas as i see this error very often in the field

Greetings

dtometzki, 2024-02-16:
That would be great

Sn0w3y, 2024-02-16:
> This could happen if the KEBA was not read from the display but from the router as described in the documentation.

Hi Sebastian,

i also want to understand. But i do not understand your saying in this sentence :D The Keba is not read from Display ? 
I guess you mean the IP?
Also the IP can be set Dynamic so the approach to read the IP from the Router is also possible instead.

Greetings

dtometzki, 2024-02-16:
Hi @sebastianasen, 

yes it can be the case. The Wlbox has one IP and the FEMS has an other IP. But the main issue is when it is in Error State. The Walbox doesnt use the settings. The result is the Walbox load the car with Full speed without regulation. 

Please do a fix for this issue. I wrote a message to service @ fenecon. 

Many Thnaks for you help
Damian

dtometzki, 2024-02-16:
Hello together,

i found the problem with fenecon. The Problem was the KEBA APP in FEMS Monitoring had the wrong IP set. 
We set it correctly and now it works. 

Perhaps the Error Message can be improved here. 

But many thanks for all the help. 

Damian

closed 2024-02-16 by dtometzki

dtometzki, 2024-02-16:
Maybe also try to set the IP to "fixed" in your Router so the KEBA does not get another IP Address assigned.

Yes i did now