OpenEMS/openems #2556: Missing implementation for  set reactive power

janwach, 2024-02-29:
### Bug Report or Feature Request (mark with an `x`)
```
- [ x ] bug report -> please search issues before submitting
- [ x ] feature request
```

### Bug description or desired functionality.

**Missing implemenation of reactive power setpoint handling**

Hello everyone,

we are currently using a FENECON Home system with a goodwe inverter for a project. We want the system to follow given power setpoints which we command using the modbus tcp/ip API. However, the system does not react to reactive power setpoints. Looking at the code for the goodwe inverter (io.openems.edge.goodwe/src/io/openems/edge/goodwe/ess/GoodWeEssImpl.java ) I noticed that it seems that the functionality is not implemented in the edge device?

Thank you in advance for any insights.

Best

Sn0w3y, 2024-02-29:
@sfeilmeier please close these issues as these are:

1. not directly openEMS related 
2. these issues only appear in this case - would have been seen in the field @fenecon i guess?

as mentioned here:

https://github.com/OpenEMS/openems/issues/2557

sfeilmeier, 2024-03-03:
@janwach: If there is an issue with or you are missing a purchased feature with a FENECON system, please get in touch with FENECON support -> https://fenecon.de/kontakt/.

For OpenEMS discussions and feature requests, we recommend the OpenEMS Community Forum -> https://community.openems.io. In fact we plan to close the Github Issues feature, to have all OpenEMS discussions in one place.

I'll close the issue here. Thanks!

closed 2024-03-03 by sfeilmeier